Ultimaker/Cura #5251: [Cura 4.0.0-BETA] Override unknown filament has no effect in app Monitor (contrary to Cura Connect)

MoonCactus, 2019-01-30:
**Application Version**
Cura 4.0.0-BETA

**Platform**
Xubuntu 18.04.1 LTS (Bionic Beaver)

**Printer**
UM3

**Steps to Reproduce**

Start a job with an "unknown" filament that was loaded manually (no NFC, that was loaded as "PLA" manually)
Slice and try to print remotely from Cura 4
Cura "MONITOR" panel tells that it "Requires configuration changes"
Click "Details" button
Click the "Override" button of the corresponding button
It has no effect

[image]

**BUT**

Go to Cura Connect
You are being told about the same issue
But there you can proceed!

[image]

Ghostkeeper, 2019-01-31:
Apparently none of the posting API calls work right now from the entire monitor. We're working on it.

MoonCactus, 2019-04-17:
I was able to override the config in Cura 4.0.0 for the last weeks, maybe this issue was addressed & can be closed?

Ghostkeeper, 2019-04-17:
Yup, I believe this was fixed. Thanks for re-testing!

closed 2019-04-17 by Ghostkeeper