Ultimaker/Cura #5253: Cura changing created size.

JoeteckTips, 2019-01-30:
I created my project using cm (center meters) and Cura brings it in using mm. Now, my printer can print 300x300x400, but my measurements are cm because its a big project. Probably will take over a day to print.

How can I tell Cura to load it as cm instead of mm? 

Thanks
JoeteckTips
www.youtube.com/joeteck

fieldOfView, 2019-01-31:
Most 3d model formats do not specify units, so an assumption must be made. The most common assumption is millimeters, so that is what Cura uses. There is no option in Cura for it to use another unit when opening a file.

PS: You meant centimeters

Ghostkeeper, 2019-01-31:
STL files don't specify any unit, but 3MF files specify that it has to be mm. If you export your project from CAD as a 3MF file then I expect that your software would convert it to mm so that you can use it in Cura without needing to manually resize it to 1000%.

Cura assumes that your project is in mm, with two exceptions:
* If the "Scale extremely small models" preference is enabled and your model is very small, Cura will resize it with factors of 10. Of course, printing a 4mm object is plausible so there is a grey area where Cura will not resize it automatically. The threshold here is 1/100th of your build volume in all dimensions.
* There is a plug-in for Cura to interpret models as being modelled in barbaric units (inches). A similar plug-in could be made for centimetres, but that doesn't exist yet as far as I know.

closed 2019-01-31 by nallath