Ultimaker/Cura #5254: run from source: python error: Segmentation fault

JockJo, 2019-01-31:
**Application Version**

the latest source code
**Platform**

ubuntu 18.04 LTS
**Printer**

**Steps to Reproduce**

i just run the cura_app.py by pycharm IDE

then report the error as follow:

Fatal Python error: Segmentation fault

Current thread 0x00007f903db5a740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 205 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 922 in create_module
  File "<frozen importlib._bootstrap>", line 560 in module_from_spec
  File "<frozen importlib._bootstrap>", line 648 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 950 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 961 in _find_and_load
  File "/home/jockjo/Cura/cura_app.py", line 129 in <module>
**Expected results**

**Additional Information**

Ghostkeeper, 2019-01-31:
It may help if you can run it in debug mode to find out what module it is trying to load that is causing this segfault.

And what happens when you try to run it from a command line?

JockJo, 2019-01-31:
thanks for your answer.
when i ran in debug mode, i got the information as follow:

```
Fatal Python error: Segmentation fault

Thread 0x00007f5df4ef3700 (most recent call first):
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 299 in wait
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 551 in wait
  File "/snap/pycharm-educational/10/helpers/pydev/pydevd.py", line 128 in _on_run
  File "/snap/pycharm-educational/10/helpers/pydev/_pydevd_bundle/pydevd_comm.py", line 320 in run
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f5df56f4700 (most recent call first):
  File "/snap/pycharm-educational/10/helpers/pydev/_pydevd_bundle/pydevd_comm.py", line 382 in _on_run
  File "/snap/pycharm-educational/10/helpers/pydev/_pydevd_bundle/pydevd_comm.py", line 320 in run
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f5df5ef5700 (most recent call first):
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 299 in wait
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/queue.py", line 173 in get
  File "/snap/pycharm-educational/10/helpers/pydev/_pydevd_bundle/pydevd_comm.py", line 459 in _on_run
  File "/snap/pycharm-educational/10/helpers/pydev/_pydevd_bundle/pydevd_comm.py", line 320 in run
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/home/jockjo/anaconda3/envs/cura/lib/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f5e00f4f740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 205 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 922 in create_module
  File "<frozen importlib._bootstrap>", line 560 in module_from_spec
  File "<frozen importlib._bootstrap>", line 648 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 950 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 961 in _find_and_load
  File "/home/jockjo/Cura/cura_app.py", line 129 in <module>
  File "/snap/pycharm-educational/10/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18 in execfile
  File "/snap/pycharm-educational/10/helpers/pydev/pydevd.py", line 1135 in run
  File "/snap/pycharm-educational/10/helpers/pydev/pydevd.py", line 1735 in main
  File "/snap/pycharm-educational/10/helpers/pydev/pydevd.py", line 1741 in <module>

Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)
```

I guess that the error occursed when the script imported module Arcus. Because when ran to here the error was produced.
```
# Workaround for a race condition on certain systems where there
# is a race condition between Arcus and PyQt. Importing Arcus
# first seems to prevent Sip from going into a state where it
# tries to create PyQt objects on a non-main thread.
import Arcus #@UnusedImport   the error was produced when ran to here
import Savitar #@UnusedImport
from cura.CuraApplication import CuraApplication

app = CuraApplication()
app.run()
```

JockJo, 2019-01-31:
> It may help if you can run it in debug mode to find out what module it is trying to load that is causing this segfault.
> 
> And what happens when you try to run it from a command line?

JockJo, 2019-01-31:
thanks for your answer.
when i ran it in debug mode, i got the output as follow.

```
(gdb) run /home/jockjo/Cura/cura_app.py
Starting program: /home/jockjo/anaconda3/envs/cura/bin/python3.6 /home/jockjo/Cura/cura_app.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:93
93	../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.

```

JockJo, 2019-02-01:
Dear Ghostkeeper ,thanks for your help.
I have solved the problem when i change the version about pyQt5

closed 2019-02-01 by Ghostkeeper