Ultimaker/Cura #5261: Can Quality Presets and Generic Materials and Definitions be updated as a Bundle?

filipgoc, 2019-02-01:
**Application Version**

n/a
**Platform**

n/a
**Printer**

n/a or IMADE3D JellyBOX 
**Steps to Reproduce**

n/a
**Actual Results**

n/a
**Expected results**

n/a
**Additional Information**

I am slightly struggling with the bundles. From all I could find I see that indeed materials and quality presets should be wrappable as a bundle. But what about definitions? And can I wrap generic profiles if I need to make an upgrade for a specific machine? (I don't think so.) 

Basically my question is whether I can add or revise a support for a particular machine wholly as a bundle instead of submitting a PR. I don't think so... but I am really not that clear on bundles. 

Thanks for talking to me a bit more about bundles. I find the docs slightly insufficient for my understanding.

filipgoc, 2019-02-01:
A related question: since bundles can wrap a plugin, is it then possible (if above is impossible) to update definitions/ generic material profiles/ quality profiles with a plugin wrapped as a bundle?

Ghostkeeper, 2019-02-04:
> From all I could find I see that indeed materials and quality presets should be wrappable as a bundle.

Yes, but only one material per package.

> But what about definitions?

> Basically my question is whether I can add or revise a support for a particular machine wholly as a bundle instead of submitting a PR. I don't think so... but I am really not that clear on bundles.

No, but that's on the planning to start supporting at some point. We want to allow people to contribute machine definitions - and all profiles belonging to one - via the contributor portal so that we don't have to distribute them only via the Cura releases. It's not in yet though and I can't say when this will happen (the plan has been there since the summer).

> And can I wrap generic profiles if I need to make an upgrade for a specific machine? (I don't think so.)

You should be able to add profiles to generic materials, if that's what you're asking. You can't update the global default profiles as they are not registered to belong to any package.

> A related question: since bundles can wrap a plugin, is it then possible (if above is impossible) to update definitions/ generic material profiles/ quality profiles with a plugin wrapped as a bundle?

Not normally. A package can either be a material package or a plug-in package, not both. However there are some hacks that can be made. After all, your plug-in can execute arbitrary Python code so it could add some materials and qualities to the configuration directory, or even to Cura's internal container registry on the fly.

filipgoc, 2019-02-04:
Thanks for the explanation!

closed 2019-02-04 by filipgoc

ChrisTerBeke, 2019-02-04:
Have a look at the Dremel3D plugin, they do some python magic to install definitions, so it kinda acts like a machine package.

filipgoc, 2019-02-04:
> python magic to install definitions, so it kinda acts like a machine package.

Awesome. 👍