Ultimaker/Cura #5263: [Request] Making support brim ignore "Brim Distance".

JohnEdwa, 2019-02-02:
10 I suggest either removing the gap between the bim and supports no matter what the "Brim Distance" is set as, implementing a "Brim Support Distance" to override this value, or combining such a behavior to one of the options such as "Support Brim" or "Brim Replaces Support", as those are quite clearly intended by the user as "Please make sure these supports stick".

---

As to why:

20 I've figured out an almost foolproof way to generate supports for tiny thing that will practically *never* be dislodged from the buildplate by combining "Support Brim", "Tower Diameter" and "Brim Replaces Support", then generating a brim wide enough to capture them all (further enhanced by adding single-layer custom supports if need be):

[image]

30 The only problem is, I don't actually want a brim with 95% of my models, which would be fixed by using "Brim Distance", if it didn't by default also add a gap between the supports and their brim, completely negating the point:

[image]

40 However, with the "Brim Replaces Support" option it is already generating the support on top of the brim if it can, so it's not like it's not combining the support and brim anyway.

[image]

50 Therefore,

GOTO 10 

:)

---

**Application Version** 4 Beta / 20190131 Master

**Platform** Win 7 64-bit

**Steps to Reproduce**
 * Generate supports while having a brim enabled
 * Modify "Brim Distance" value

**Actual Results**
 Cura will pull back the brim from the supports.

**Expected results**
 Cura should create a gap only between the model and the brim, not the supports and the brim.

Ghostkeeper, 2019-02-04:
So why don't you try removing the normal brim and setting the support brim to have a much larger brim width?

Alternatively, set the bed adhesion type to Skirt, then set the minimum skirt length very high so that it behaves as a brim.

JohnEdwa, 2019-02-04:
> So why don't you try removing the normal brim and setting the support brim to have a much larger brim width?

Because "Support Brim" only does the brim inside the original support footprint replacing the first layer, and doesn't actually work like the regular brim by creating extra around the part. This means you'd also have to massively crank the "Tower Diameter" or Cura will still try to create tall skinny supports with tiny footprints that will not adhere to the bed properly. But doing so increases material consumption by a lot more than just printing brim around them.

> Alternatively, set the bed adhesion type to Skirt, then set the minimum skirt length very high so that it behaves as a brim.

This only works if you set the "Skirt Distance" to 0, as otherwise even with a 0.1mm gap the skirt starts from the supports outwards not from the actual part, and will miss any supports closer in. And if you do set it to 0, it's attached to the original part like a brim with no gap and we are back to square one.

[image]

ungutknut, 2020-05-26:
This also is an issue for me. 

Having something like a real support brim would be a nice thing, but currently (4.6.0) I don't see an option to achieve this. As you can see in my screenshot, cura is ignoring the desired amount of support brim lines (13) and just creates a brim that doesn't exceed the supported geometrys footprint... which isn't actually a brim. If this is intented then this feature should be renamed.

Ghostkeepers suggestions don't address the issue so I'm not sure if the problem has been understood completely. Because at least to me it definitely looks like support brims are broken.

[image]

Ghostkeeper, 2020-05-28:
Indeed we are talking about two different things here. You seem to be talking about making the Support Brim go around the outside of the support, i.e. [this feature request](https://github.com/Ultimaker/Cura/issues/6447), while I think JohnEdwa as talking about removing the Brim Distance in places where the normal brim goes around the support rather than around the model.

ungutknut, 2020-05-28:
Ok, thanks for correcting me. However I think a real support brim that behaves like normal brim - but only for support - would also be the solution to JohnEdwa's problem.

JohnEdwa, 2020-05-28:
My side of this seems to already have been fixed, no idea when that was done though. But as such, I'll close this one.

I'm not sure if the support brim should always go to the outside, but I do think that having the supports be able to generate brims on the outside as well when they don't fit on the inside would be a good improvement (so a 10mm support brim on a 5mm cylinder of support would first fill the inside, then "bleed" to the outside to make sure the total brim is 20mm in diameter as requested), but that should go to another ticket.

But for now, using a brim with a small gap will ensure the supports get a proper brim, yet it doesn't stick to the actual model. Wastes a little plastic and time, but it's good enough.
[image]

closed 2020-05-28 by JohnEdwa