Ultimaker/Cura #5264: Can't see shape of part in solid or layer view if material color is black.

Nandox7, 2019-02-02:
**Application Version**
3.6 & 4.0

**Platform**
Windws 10

**Steps to Reproduce**
Import part with active material of black color.

**Actual Results**
Can't see the shape of the part or preview. Is all a black shadow.

**Expected results**
Be able to see the part shape.

**Additional Information**
(screenshots added)
[image]
[image]

Ghostkeeper, 2019-02-04:
Can't reproduce this. At least, not as bad as your screenshots.
[image]
I wonder how your first screenshot got so dark.

Nandox7, 2019-02-04:
The first screenshot is the layer view. (added some labels now to the images)
Try that on your side and see if the same happens.

Ghostkeeper, 2019-02-04:
Ah yes, I didn't read properly. Here you go.

[image]

Darker indeed. We might be able to do something about that. But it's still much more visible than yours. Strange. And you also seem to have OpenGL 4 because your layer view is modern.