Ultimaker/Cura #5265: [CURA 4.0.0 BETA] Black screen after 2 minutes with Ryzen 5 1600 and win10

fanfan7110, 2019-02-02:
**Application Version**
<CUra 4.0>

**Platform**
<Windows 10->

**Printer**
<Alphawise U20

**Steps to Reproduce**
<Just install in a ryzen5 >

**Actual Results**
<Black screen->

fieldOfView, 2019-02-02:
Can you check if the memory use of Cura increases in those 1 to 2 minutes (using the windows Task Manager)? If that does not result in something that explains the lockup, checking temperatures may also be interesting (but you'll have to find a tool for that yourself).

Ghostkeeper, 2019-02-04:
Google says to look at power settings, such as this: https://youtu.be/bheUmk2qEp4?t=82

fanfan7110, 2019-02-04:
So, there is no memory increase except when it slice..  No excessive temperature. 
For windows setting and bios i'll see that.

Ghostkeeper, 2019-02-19:
We have a similar report on Ultimaker's forum now: https://community.ultimaker.com/topic/26654-cura-crashes-my-entire-computer/

fanfan7110, 2019-02-19:
yes, it's exactly the same problem. and with two ryzen.... it's not a coincidence

fieldOfView, 2019-02-19:
Since none of the developers have a Ryzen system, we really need your feedback to know if the proposed power settings/bios changes fix the issue.

fanfan7110, 2019-02-19:
Absolutly not. This setting have no effect.

janrg, 2019-02-19:
Unfortunately, my MB's UEFI (ASUS STRIX X-470-F, UEFI Revision 4018) doesn't seem to have an option for disabling C-states. I've changed the power plan now and will try to provoke the issue into occurring. Will report back.

janrg, 2019-02-19:
Sadly, that did not do the trick. I left Cura open and while I was using the Computer it worked fine, but when I stepped away for 10 minutes, it locked up again.

janrg, 2019-02-21:
Oh, I should point out I'm actually running 3.6, not 4...

Ghostkeeper, 2019-02-22:
It's very hard to say something about this issue :disappointed: We don't know what's causing it or why it only happens with Ryzen.

Cura's only binary code under our source control is in CuraEngine/libArcus but since this happens during idle that's not running. The rest of Cura's binary code is in our dependencies such as Qt, CPython, etc. so then you would expect this to happen with other applications as well that depend on the same dependencies.

Maybe this also happens with other Qt-based applications, such as FileZilla or OpenSCAD?

fanfan7110, 2019-02-22:
in fact no, i use Fillezilla all days and there is no problem with it. 
To be precise, no software installed in my machine reproduces the problem found on cura.
I use daily software like filezilla, fusion360, meshmixer, Zbrush without any worries.
Still yesterday, I made the settings of cura 4.0 and during keyboard editing parameter entry the screen became black and it was finished.
Note that often, after reboot, Cura forces reset the settings to zero to return to its basic installation state (if that can help). So you have to tell again what is the printer etc.

Ghostkeeper, 2019-02-22:
That means that the configuration files got corrupt. It may have happened during saving of profiles.

robbartley, 2019-02-24:
Hello I am also having this problem

CPU: Ryzen 5  2600
GPU: Nvidia GTX 1070

I think there must be an issue with cura vs AMD CPUs

Tibdarko, 2019-02-26:
Same with :
CPU: Ryzen 7 2700X
GPU: Nvidia GTX 1080
RAM : 16 Go (2 x 8 Go), 3000 MHz, CAS 15
Win 10

fanfan7110, 2019-02-26:
See the list of users that extends, it is no longer a simple bug strictly local, but a problem operating Cura with a Ryzen processor .. The list of blocked users may well lengthen a little every day :)

Ghostkeeper, 2019-02-27:
On [this thread](https://www.reddit.com/r/AMDHelp/comments/804fv7/ryzen_7_random_black_screen/) they seem to arrive at it being a problem of the CPU not getting enough power. They also indicate that it often happens during idle time just after a CPU-intensive task, which is very typical for Cura's usage pattern.

fanfan7110, 2019-02-27:
It can not be that, because the bug appears even if we did not even start loading a model.
For example, launch Cura and  just  wait a few minutes for the screen to turn black.
Or just access the cura's settings for cut and block the pc. So unless you consider that access to cura parameters is an overload of the processor (in this case you have to redo all programming software, lol) it is impossible that it comes from what you said.

Ghostkeeper, 2019-02-27:
Launching Cura is a pretty CPU-intensive task :P

fieldOfView, 2019-02-28:
Do you have the "Aura" or "AI Suite" mentioned in the thread above installed?
There are more reports of black screen issues connected with these packages:
https://forums.tomshardware.com/threads/ryzen-27700x-black-screen-issue.3393849/
https://community.amd.com/thread/231672

robbartley, 2019-02-28:
just out of interest, I noticed my GPU temperature (GTX 1070) was idling REALLY high

might be worth checking if your fans are spinning and if your fan curves are correct

it feels somewhat like the GPU maxing out on temperature and shutting down which causes a cascading issue through the rest of the system

robbartley, 2019-02-28:
update:  issue still persists.

Although I think it's prudent for everyone to check their GPU is working as intended.

just uninstalled AURA as per fieldOfView's suggestion, already had AI suite uninstalled

adamheinmiller, 2019-03-03:
Ryzen 2700 X
Nvidia GTX 1060
Cura 3.6

I'm also experiencing this issue.  My system is otherwise completely stable.  I have the same symptoms as everyone else - system goes dead without warning, have to physically cut power to get it to start back up.

It occurs whether I'm actively using Cura or have it in the background.  It usually happens 5-8 minutes after starting Cura.  Confirmed it's not CPU/GPU thermals - they are near idle temps if I open Cura and let it sit until a crash.  I do have Ryzen Master installed but nothing is customized.  I've also tried using Process Lasso to restrict Cura to a single core, but it has no effect.

fieldOfView, 2019-03-04:
It would be very helpful if anyone with a spare partition could do a clean installation of Windows 10 without installing AURA and AI Suite, and see if it reproduces there. From what I have read in the linked articles, uninstalling does not fix it (leaves something behind?).

fanfan7110, 2019-03-04:
> It would be very helpful if anyone with a spare partition could do a clean installation of Windows 10 without installing AURA and AI Suite, and see if it reproduces there. From what I have read in the linked articles, uninstalling does not fix it (leaves something behind?).

Already done and explained in the first post. Even after full formatting and a clean reinstallation without anything else, as soon as Cura is reinstalled and started up the problem is present. In fact, when I reformatted my PC, the crash even appeared about 10 seconds after the first start of Cura 4. The hardware was a Ryzen 5 1600 with Asus Prime X370 Pro card and 16 GB of RAM.
I changed yesterday for a Ryzen 7 2700X with motherboard
Gigabyte GA-B450M-DS3H. For now no blocking, but as it worked sometimes .. I will see with time

robbartley, 2019-03-04:
Uninstalling AURA and having already uninstalled AI seems to have worked for me.  Have used CURA 3.6,0 extensively in recent days and no random blackouts.

ROG STRIX B450-F
Ryzen 5 2600
EVGA GTX 1070 SC2

NZXT CAM and EVGA Precision X1 installed

janrg, 2019-03-04:
I have neither AURA nor AI installed...

robbartley, 2019-03-04:
List your build specs?

Mobo
Cpu
Gpu psu

janrg, 2019-03-04:
Oh sorry, I thought I had already listed that here, but I only did in the Cura forum.
ASUS STRIX X470-F
R7 2700X
EVGA RTX 2080 XC
Seasonic 660 Platinum XP2

adamheinmiller, 2019-03-05:
> It would be very helpful if anyone with a spare partition could do a clean installation of Windows 10 without installing AURA and AI Suite, and see if it reproduces there. From what I have read in the linked articles, uninstalling does not fix it (leaves something behind?).

I've never had either utility installed on my system.

I didn't list my motherboard in my prior post.  I have the ASUS Strix X470-F.

robbartley, 2019-03-05:
So the issue appears to have returned when I just left CURA idle for about 30minutes...

Very very odd

janrg, 2019-03-05:
It does appear though that aside from the Ryzen CPU, the common factor between all of us is that we have ASUS motherboards...

robbartley, 2019-03-05:
I am potentially swapping mine to an MSI soon as the ASUS AM4 boards aparently have poor VRM so this could be a contributing factor

I read the same issue (minus the cura part) on an amazon review also, so CURA may be coincidental rather than causational

adamheinmiller, 2019-03-05:
> It does appear though that aside from the Ryzen CPU, the common factor between all of us is that we have ASUS motherboards...

> I am potentially swapping mine to an MSI soon as the ASUS AM4 boards aparently have poor VRM so this could be a contributing factor
> 
> I read the same issue (minus the cura part) on an amazon review also, so CURA may be coincidental rather than causational

We can't rule it out certainly, but I think another element here is we all have otherwise stable systems, and ASUS is a very popular board manufacturer - it was the only one available when I purchased my Ryzen, so a small sample size might make this a coincidence.

I've left my system running 24+ hours, pedal to the metal, doing stress tests on OC's.  I've also had it running multi-day video encoding jobs (maxed CPU/GPU, high memory + disk IO) and it has been eminently stable.  I think any fundamental stability issues with the board would be exposed in those extreme workloads.  Cura, or something it does, seems unique in its ability to brown-note my system.

janrg, 2019-03-05:
Just wondering do you ever put your Ryzen PC on standby? Thinking back to when I first set mine up, I used to get similar problems sometimes when I set it to standby, which is why I haven't done that in a long time. I think perhaps cura is triggering some underlying issue in the ASUS motherboards...

rxdpunk, 2019-04-08:
Let me add my system to  this list. 
Ryzen 7 1700 4ghz
16gb ddr 4 3000mhz
Gtx 1070

I get the same exact crash as reported by other users. In fact I didn't realize what was causing my crash for awhile. I ended up removing all over clocking from my system.  Tinkered with power states. Removed all system standby. Even bought a new monitor because I thought it was failing lol.

fanfan7110, 2019-04-08:
For information, as I said above, I switched to a Ryzen 7 2700X with Gigabyte GA-B450M-DS3H motherboard. CEla is now several days (see weeks) and with this new equipment I did not have any blocking or bug .. Would it be fair with version 1 of ryzen?

rxdpunk, 2019-04-10:
I did drive format and fresh install of windows 10 home yesterday.  So far cura 4 runs without crashing the system anymore.

TheFluffyWalrus, 2019-04-22:
I am also having the same issue also all the others having the same issue also seem to be running Nvidia GPUs... 

Ryzen 7 2700x
Asus Prime x470 Pro
32gb Corsair 3000mhz
Nvidia 2080RTX

AstraelOrannis, 2019-04-24:
Also having this problem.

Cura 4.0.0
Ryzen 2700X
Nvidia RTX 2080

32GB RAM
ASUS Strix X470-F motherboard

CPU, VRM and GPU are liquid cooled, temps are 50C or less.

I pulled my OC off the CPU and GPU still am getting blackscreened after about 30 minutes in CURA.

BIOS are updated and I have checked CPU voltage and made sure there isn't a power problem.

Ghostkeeper, 2019-04-25:
So far we've seen it with all sorts of Ryzen chips:
* 7-2700X (most common)
* 5-2600
* 5-1600
* 7-1700

Maybe it only happens when the PSU is not strong enough?

rxdpunk, 2019-04-25:
I have 850w power supply happens under idle

AstraelOrannis, 2019-04-25:
750w 80+ gold PSU
Also happens under idle

TheFluffyWalrus, 2019-04-25:
Thermaltake Smart 750w here... def does it on idle doesn't matter when it's completely random sometimes it crashes a few minutes in sometimes 30+ one time it actually stayed running after overnight when I went to bed and forgot to shut the machine down.

fanfan7110, 2019-04-25:
Hummm ... Have you see that all your motherboard are ASUS ???

fanfan7110, 2019-04-25:
i tell that because since i have change my asus motherboard with a gigabyte, i don't have bug anymore

rxdpunk, 2019-04-25:
If that is the the case I dont understand.  What is happening between this one piece of software and a motherboard?  I can push this PC to play destiny 2 maxed out for 8hrs straight.

fanfan7110, 2019-04-25:
i know, i know (i'm a destiny 2 player's too) but it's a fact .. i don't have bug anymore

robbartley, 2019-04-25:
It isn't a matter of how high you can push your PC or PSU's being undersized

to me the common issue here is ASUS motherboards with Ryzen chipsets.  Could it be down to ASUS typically not having very good VRMs on their boards?

Also I find it bizarre that I ONLY experience this issue with CURA, no other program does it

fanfan7110, 2019-04-25:
It's right and I think that's the case for everyone here. .Only Cura 4 create this problem and no other software... 
In fact, I don't understand why the Cura developers team doesn't rent or borrow just a PC under Asus / Ryzen to see the logs of the development software.

robbartley, 2019-04-25:
I had the problem with cura 3.6.0 and cura 4

For the amount of money ultimaker make, it feels like the €500 it would take to build a decent ASUS Ryzen PC to test this is trivial. Especially considering how widespread Ryzen is becoming

Ghostkeeper, 2019-04-29:
For Ultimaker it's always been a matter of not having enough people to do the work rather than enough money or tools.

Not all Ryzen computers reproduce this though. One of our colleagues just got a Ryzen as well with ASUS motherboard, but doesn't see it happen (so far).

I would also have no idea where to look for a fix :confused: It's not like Python has an instruction that says "please freeze the CPU and make the screen go black" that accidentally gets triggered in our code due to a typo or something. It'll be either a bug in CPython, Qt or in GCC together with these Ryzen chips, and I suspect it's some bug in the power control of that motherboard which triggers on the type of power behaviour that Cura has, i.e. bursts of high CPU usage (like the internet says).

TheFluffyWalrus, 2019-04-29:
So since I can reproduce this what can I do to help you fix the problems for other people?

Ghostkeeper, 2019-04-29:
Maybe you could try putting Cura and CuraEngine's process on a lower priority?

TheFluffyWalrus, 2019-04-29:
So let's say prime95 to eat up CPU? While cura is on low priority? And putting cure on low priority I do that in task manager correct it's been awhile since I've set a priority for an application.

TheFluffyWalrus, 2019-04-30:
Cura and the slicing engine both on low priority with prime95 eating everything so far so good... it is 12:31AM EST will leave this running while I sleep.... @Ghostkeeper Another thing I noticed about everyone crashing is we all have NVIDIA cards as well. I don't think its effecting people with ATI cards.... Could it also be GPU related?

Ghostkeeper, 2019-04-30:
Do you have ASUS AI Suite installed? According to [this thread](https://www.reddit.com/r/Amd/comments/8ujrii/psa_asus_b350f_crashes_with_ai_suite/) and [this post](https://community.amd.com/thread/231672) it can crash the GPU. It's been mentioned before [here](https://github.com/Ultimaker/Cura/issues/5265#issuecomment-468178699)

This guy with a similar issue mentions that he had installed MSI command centre and it pops up a message saying that his voltages are wrong: [thread](https://forum-en.msi.com/index.php?topic=291762.0)

I also see a lot of threads online where they ask the user to upgrade the BIOS on the motherboard or the BIOS on the GPU. And a lot of people complaining that ASUS quality control is terrible...

TheFluffyWalrus, 2019-04-30:
I dont have ASUS AI suite installed... I will update my BIOS and and see if there is a BIOS update for my Founders Edition RTX then do the test again and report back.

rxdpunk, 2019-04-30:
I clean installed no AI suite current bios on everything still happens

Ghostkeeper, 2019-04-30:
Maybe something else you could try is to install a very old version of Cura: http://dulek.net/work/Cura-2.1.2-win64.exe
Installing this should not influence your modern Cura installation unless you install it over the same folder (but by default they are installed to different folders).

[347 more replies]