Ultimaker/Cura #5269: cura 4.0.0 (beta) systerm slow on win 10m pro

mickletown, 2019-02-04:
4.0.0 beta

**Platform**
dell e6410 laptop with  win 10 pro

**Printer**
prusa i3
**Steps to Reproduce**
running 4.0.0 beta with 3.5 running in background

**Actual Results**
 system slows to near stop

**Expected results**

**Additional Information**

fieldOfView, 2019-02-04:
Didn't 3.5 have a memory leak that would be triggered for some people when Cura was left running in the background? As in: it could be 3.5 causing the system slowdown (excessive virtual memory paging), not 4.0

Ghostkeeper, 2019-02-04:
Ah yeah. If connected with a camera stream, wasn't it? That could happen with Prusa i3 as well if Octoprint was providing the stream source.

fieldOfView, 2019-02-04:
In case OctoPrint Connection plugin is part of the equation, @mickletown may want to update the plugin through the Marketplace in Cura 3.5. If that does not help, uninstalling the plugin, restarting Cura, and reinstalling it (and restarting Cura again) may help.

closed 2020-06-20 by no-response[bot]