Ultimaker/Cura #5272: [3.6.0] Please stop detecting file changes in real time

FlailAway, 2019-02-04:
Gotta say I am loving Cura, but the file handling in Ubuntu/Mint at least, needs a little work to make it more user-friendly. But, back to the bug-like issue.

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

Using Workspaces: Cura running in Workspace-1. FreeCAD running in Workspace-2. Load .obj file from Workspace-2 into Cura, Slice and "Save to File." Leave as is, go to FreeCAD Workspace, make changes and save. Come back to Cura Workspace and the attached pic says it all. 

**Actual Results**

Multiple panels that must be cleared before Cura is alive again. Must be removed by clicking "X" in the lowest panel until all are gone. Make ten change/save then ten panels must be removed.

**Expected results**

It does happen, Cura comes back to being usable after clearing all the "RELOAD..." panels.

**Additional Information**

A picture paints a thousand words. Cura needs to stop monitoring the currently open file to request "RELOAD..."

[image]

smartavionics, 2019-02-04:
This bug has been around for quite a while (https://github.com/Ultimaker/Cura/issues/3936). They don't seem to be worried about fixing it.

smartavionics, 2019-02-04:
You know, thinking about this some more. I don't know why they have this feature at all. Given that the user has probably just updated the model file (STL/OBJ) by exporting it from their drawing package.

It's easy to disable these messages with a little edit to the code but as you are using the AppImage I don't know if you can get to the Python easily. Anyway, here's a patch that will stop these messages appearing. They have already disabled this feature for Windows as it causes crashes. It seems to me that it would be much better if it was disabled completely for all platforms.

````
diff --git a/UM/Scene/Scene.py b/UM/Scene/Scene.py
index d09b33540..cccc640c7 100644
--- a/UM/Scene/Scene.py
+++ b/UM/Scene/Scene.py
@@ -145,14 +145,14 @@ class Scene:
     #   \param file_path The path to the file that must be watched.
     def addWatchedFile(self, file_path: str) -> None:
         # The QT 5.10.0 issue, only on Windows. Cura crashes after loading a stl file from USB/sd-card/Cloud-based drive
-        if not Platform.isWindows():
+        if False and not Platform.isWindows():
             self._file_watcher.addPath(file_path)
 
     ##  Remove a file so that it will no longer be watched for changes.
     #   \param file_path The path to the file that must no longer be watched.
     def removeWatchedFile(self, file_path: str) -> None:
         # The QT 5.10.0 issue, only on Windows. Cura crashes after loading a stl file from USB/sd-card/Cloud-based drive
-        if not Platform.isWindows():
+        if False and not Platform.isWindows():
             self._file_watcher.removePath(file_path)
 
     ##  Triggered whenever a file is changed that we currently have loaded.
````

smartavionics, 2019-02-04:
I don't know Python so maybe it's possible to simply omit the bodies of those functions completely.

FlailAway, 2019-02-04:
OK, thanks, but what do I do with the Patch? I just did some searching and cannot find anything on applying patches to Cura.

A bit bizarre that this issue has been removed from windows version, but the poor old Hay-seeds using Linux still must suffer. :)

smartavionics, 2019-02-04:
Sorry, I don't know how the file in the AppImage can be altered. Perhaps it's possible to unpack the AppImage, edit the file and either pack the AppImage again or just execute Cura from the unpacked files. I don't have any experience of doing that.

smartavionics, 2019-02-05:
> I don't know Python so maybe it's possible to simply omit the bodies of those functions completely.

No, that doesn't work so the simplest fix is to insert a return statement as the first line of those two functions (addWatchedFile() and removeWatchedFile()).

Ghostkeeper, 2019-02-05:
I don't use this feature very often but when I do, it's a bit annoying.

The simplest proper fix would be to create only one message and just make it visible and change the text every time it needs to appear. I looked at it in between stuff a while ago, but that last part needs a minor refactor to allow changing the text on the fly, iirc.

smartavionics, 2019-02-05:
But given that the feature has been disabled on Windows anyway, why not just remove it completely?

fieldOfView, 2019-02-05:
> that last part needs a minor refactor to allow changing the text on the fly, iirc.

Just close it and recreate it with a new text. Much less annoying than having multiple messages stacked on top of eachother.

thopiekar, 2019-02-05:
No. I contributed the feature, yes, but it is not part of any plugin. I contributed directly to Cura/Uranium, I think.

thopiekar, 2019-02-05:
@smartavionics The feature is basically not bad. It just needs some tuning like @fieldOfView suggested.
The comment says that it only happens on 5.10.0 (or the 5.10 series?). Why not adding a check, which Qt version we are running at?
Think that would make sense, so as soon as we are running on more recent Qt, then it should work.

smartavionics, 2019-02-05:
You may be right but I've given up with it and now it is disabled. For me, the feature is just irksome and provides zero benefit.

FlailAway, 2019-02-06:
> For me, the feature is just irksome and provides zero benefit.

It is irksome only when it does the multiple things. I think it is a good thing, but it needs taming. :)

`if (!panel.open) {
  panel.show;
}
`
So, where does that leave me with the appimage?

nicknugget, 2019-02-12:
Yeah, it's pretty damn annoying. I also use cura via AppImage and would be very happy if this gets fixed. I like the idea behind this feature. But right now it's just broken.

Ghostkeeper, 2019-02-13:
We fixed this for version 4.0: https://github.com/Ultimaker/Uranium/commit/1ec5023152977ec7099508302e5915bce7f7804d

It now only shows the message once.

closed 2019-02-13 by Ghostkeeper