Ultimaker/Cura #5275: Error when slicing a circle one string thick

Asger-P, 2019-02-05:
**Application Version**

3.6 and 4.0 beta

**Platform**

Winsows 7 pro

**Printer**

[Test100mmCirkel.zip](https://github.com/Ultimaker/Cura/files/2833798/Test100mmCirkel.zip)

**Steps to Reproduce**
Slice the file with a 1mm extruder 

**Actual Results**
You get this result:
[image]

**Expected results**
[image]

**Additional Information**
the sliced part is a ring with outer r=50 inner r=48.99 height = 5 all millimetre
I can understand that it happens because of the facets of the surface, but shouldn't it be the same all the way and not happen on only half the print.

Thanks you for creating a great product.
Best regards
Asger

rburema, 2019-02-07:
Hi @Asger-P,

> I can understand that it happens because of the facets of the surface, but shouldn't it be the same all the way and not happen on only half the print.

While I was having a closer look, it seems that the facets at the bottom 'line up' with each other, creating a more even band, as opposed to those at the top, where the inset of the inner circle is aligned with the flat facet of the outer one, making for a more uneven band.

cheers, Remco.

GregValiant, 2026-09-07:
I'm cleaning up these old reports and requests.  The model slices fine in 5.13.0.  I'll go ahead and close this.

closed 2026-09-07 by GregValiant