Ultimaker/Cura #5282: [FR] Smarter prime towers

paukstelis, 2019-02-07:
I really like the heat up/cool down features of Cura that proceed a tool change. I am using it extensively for a independent 5 tool printer. I do see some strange behaviours with the prime tower that I think should be explored. In the current scenario (using 4.0.0) , if I do a two color print with tools 0 and 1, and tool 0 ends up not being needed for a large part of the print in Z, it still requires a tool change back from 1 to 0 at essentially every layer to add to the prime tower. The more logical scenario would that if tool 0 is not used on that layer, tool 1 would provide both prime tower outlines to avoid the costly (in time) tool change.

Ghostkeeper, 2019-02-07:
I'll refer to my answer here: https://github.com/Ultimaker/Cura/issues/4686#issuecomment-459673190

> The whole point of this new prime tower is to make the tower printed by only one extruder so that there is not any border in the height of the tower. If the tower gets printed by multiple extruders, the tower tends to become very weak because the two materials don't adhere well to each other.

paukstelis, 2019-02-07:
I see your point, but if the two materials have similar properties then all you have is a waste of time (and an additional waste of time if you have minimum layer time turned on since it slows down the prime tower for the outer extruder).  If Cura can already detect that materials may have different properties (which it sounds like you are saying in the other thread) then surely it would be able to know that two materials have the same properties and that this tool change is not required.

rburema, 2019-02-08:
Consider also that different colors of PLA (or another material) may have (slightly) different properties, which raises the question: When is a material similar enough?

Also, I think a key part of that answer is:

> ... but I don't see an easy way to make that configurable that is also clear to users.

I think that it's a bad idea in general to switch behaviours from underneath the user without them having some way to influence that. We could introduce a boolean option, but then the measure of similarity is still an issue.

Lastly, it also makes it less testable, since a new permutation is introduced _that occurs under a vaguely defined set of criterea_ (see above) that needs to be tested for every change to the prime-tower.

paukstelis, 2019-02-08:
I think a boolean setting that basically dictates that all materials have the same adhesion with respect to the prime tower would be enough for a user. Again, I get the motivation, but it is clearly not a behavior you would want in all cases. 

I'm going to try and not just be a complainer. I've looked through CuraEngine and I'll try to generate a PR that would give what I think might be a workable solution without any fundamental changes to the prime tower.

Ghostkeeper, 2019-02-12:
There is the additional consideration too that we'd then need to maintain two implementations.

NMe84, 2020-03-04:
I was referred here from the issue mentioned above. In addition to the additional time required for obsolete tool changes, there is also the issue of this approach wasting filament. Take this [traffic cone](https://www.thingiverse.com/thing:21773) for instance. Ideally you'd have one tool change on layer 1 for the single ring of white it prints on the buildplate, then another change back to red, followed by four more tool changes to print the white rings. Instead, it switches to red and back to white for every single layer just to print the outer shell of the prime tower. Which takes a lot of time, but it also means there is less surface on the prime tower that can be used to purge the red color out and get it to print clean white instead of pink.

Just because of this "outer shell" approach to prime towers I have to make the tower much larger to make sure that each color has enough surface area to purge the old color.

I get where you're coming from in wanting to support this new way of doing prime towers but it really mostly has advantages for printers with more than two extruders. With two, you'd never need to purge more than one color per layer. I feel we should at least have the option to use a solid purge block as it used to be in Cura 3.5 originally if that makes more sense for us. Or if that's too complicated, the option to use the active extruder to build the outer shell for layers that do not require the outer layer's assigned extruder.

Ghostkeeper, 2020-03-09:
With the solid block in 3.5 we couldn't support >2 extruders, as that would require multiple prime towers.

I think that the problem really stems from how the prime tower was designed:
* For printers with multiple nozzles, where each filament is assigned its own nozzle (no colour blending issues).
* To be robust for prints with multiple different types of plastic, e.g. PLA and Nylon. With this set-up, the materials not sticking is much more of a problem than when all extruders use the same material type.

I'd also like to have a solution that works for all set-ups, but I can't justify spending time on implementing it and I need the solution to not complicate the code too much. So if you know of a way to implement this, please provide a solution.

NMe84, 2020-03-09:
> So if you know of a way to implement this, please provide a solution.

You already know if there are one, two or more extruders. I'd leave it up to the user to select if they want a traditional prime tower or the prime bucket you designed now with a checkbox that you either hide or disable if the printer profile doesn't have exactly two extruders. Or maybe a dropdown that lets you select a priming strategy, so that you could potentially add more at a later point. And yeah, that does mean you'd have to support both of the current solutions, but let's face it: the way it is now is pretty terrible for the most common type of multi-extruder printer: the kind that only has two extruders.

Alternatively just giving the user the option to use the active extruder to build the outer shell of the prime tower if the next layer doesn't require a tool change would already help a lot too, and that shouldn't be more than a couple of extra checks. As long as the corresponding tooltip warns the user about consequences for adhesion when using multiple materials that seems to me like a valid addition.

Ghostkeeper, 2020-03-12:
The first solution is really not a solution we're willing to accept. We will always prefer to have a single implementation of a functionality. If you need the old implementation you are free to use the old Cura version, but we won't keep dragging all old implementations of every feature along with us in our source code. This becomes impossible to maintain. We found the current solution to work better (and Ultimaker has no printers with 3 or more extruders), so I disagree that the current solution is pretty terrible for printers with 2 extruders.

Allowing switching which extruder gets used for the outer shell is more in the direction of a solution I'm looking for. If you're looking into this, please take care that the Prime Tower Minimum Volume is maintained correctly. That could still bite into the feasibility of this feature. The order of the shells is currently determined by the Adhesion Tendency setting (which is hidden in the GUI).

NMe84, 2020-03-12:
I don't agree that these two should be iterations of the same feature. Prime towers and prime buckets (let's call your current solution that to make distinguishing the solutions easier) are two different solutions to the same problem in the same way that rafts and brims are both solutions to the bed adhesion problem. They both still have their own use cases. The prime tower as it used to be was better for same-material multi-color prints on dual extruder printers simply because you had more control over how much material gets purged. Now it's this abstract volume number where the amount of material you need to purge vastly differs per color. I won't press the matter any further if you really don't want to consider adding the old tower functionality back but I personally feel it's as valid a distinction as the one between brims and rafts, just a less commonly required one since obviously most printers only have a single extruder.

As for the second paragraph: I am a developer so I could look into this myself and make a PR but seeing as I'm swamped with other things right now I don't think that would be any time soon. Hopefully someone else has more time than I do. If not, hopefully I can make some time later this year when my schedule clears.

On a more theoretical level: I'm not sure the minimum volume per extruder makes much sense if different extruders could print the outer shell. Whatever amount would normally be printed with whatever extruder got assigned the outer shell should be printed with the currently active extruder instead to make sure that the layer(s) above it have enough material to build on. The minimum volume setting for the first extruder should be used to determine the volume of the outer column regardless of extruder instead; if this hypothetical new feature is selected of course.

Ghostkeeper, 2020-03-17:
The minimum volume was indeed designed just to purge enough material to prime, not to ensure a certain rigidity of the outer shell. In our experience, once the shell of the prime tower is printed without material changes, the shell won't break any more. In practice, there is little difference to stability with more shell perimeters as the prime tower's width is the most major influence on the prime tower's stability.

NMe84, 2020-03-17:
How does that work actually? Volume seems a weird metric to me. I'm not interested in how heavy my purge block is, I just want to make sure that per layer that needs purging it extrudes enough material to purge colors. Would it not make more sense to define surface area or even extrusion distance per layer than to define volumes? I feel like it would be easier on everyone if we could define how much material we want to purge per extruder per tool change and to have Cura then figure out how big the prime tower would need to be.

Obviously a slightly different topic than this issue addresses but I hope you'll indulge me anyway.

Ghostkeeper, 2020-03-19:
> Volume seems a weird metric to me.

The idea is that there is a certain volume of old material in the nozzle chamber that needs to be expunged. It doesn't exactly work like that since the materials mix inside the nozzle chamber, but it's the best metric I can think of. Basically exactly what you're saying: "I just want to make sure that per layer that needs purging it extrudes enough material to purge colors". Amount of material could be defined as weight, volume or filament length. We chose volume, because indeed the weight is not interesting and extrusion distance makes people think of the feeder while it really depends on the size of the nozzle chamber and the viscosity of the material.

> I feel like it would be easier on everyone if we could define how much material we want to purge per extruder per tool change and to have Cura then figure out how big the prime tower would need to be.

This is precisely what the setting is trying to do. It's an amount of material that needs to be purged for each filament change. Cura then makes sure that it draws enough rings in the tower to purge that amount of filament. So how many rings you get will depend on the layer height, line width and tower radius as well.

Alexaandre, 2020-05-16:
hi, why cant be the rings of prime tower printed on the top of each others ? i mean, the second color doesnt need to be primed at each layers ... I tried to print that traffic cone as well, then when i saw the huuuuge waste of material being used it sacred me, how can it be done?

Ghostkeeper, 2020-05-19:
I'll refer to my answer here: https://github.com/Ultimaker/Cura/issues/4686#issuecomment-459673190

> The whole point of this new prime tower is to make the tower printed by only one extruder so that there is not any border in the height of the tower. If the tower gets printed by multiple extruders, the tower tends to become very weak because the two materials don't adhere well to each other. I agree that it's a trade-off between reliability and printing speed, but I don't see an easy way to make that configurable that is also clear to users.

Alexaandre, 2020-05-19:
Ok, i see now why it's printed like this, i like the idea of the purge bucket, what if the shell of the prime tower would be printed in "auto prime calculation mode" like universal xxx mm3 prime volume (i dont know if it can be defined, and how)  and then at each tool change or nozzle switch, (before the toolchange) the nozzle goes to the prime tower, then prime filament directly in the constructed bucket in a static motion, just ooze on the bucked instead of priming in the print then when prime is finnished, the nozzle wipes the oozed material that left on the side of the bucket on an randomized position each time ? i dont know of you see what i mean, the problem is that if there is too much toolchange, the bucket will be full at some points .

Ghostkeeper, 2020-05-22:
> Ok, i see now why it's printed like this, i like the idea of the purge bucket, what if the shell of the prime tower would be printed in "auto prime calculation mode" like universal xxx mm3 prime volume

This is what Cura is currently doing, actually. The Prime Tower Minimum Volume setting is the xxx mm³ prime volume you're talking about. It does round that up to complete the circular prime tower though.

> As i was writing this, is it not possible to make the nozzle goes on the prime tower before each toolchange ?

Normally you want to prime just BEFORE printing, rather than just AFTER printing. So it needs to go to the prime tower just AFTER the tool change. Printer definitions can define where it makes the tool change though, and it makes sense to do that over the prime tower.

> and then at each tool change or nozzle switch, (before the toolchange) the nozzle goes to the prime tower, then prime filament directly in the constructed bucket in a static motion, just ooze on the bucked instead of priming in circular motion on the prome tower then when prime is finnished, the nozzle wipes the oozed material that left on the side of the bucket on an randomized position each time ?

We tried this, but if you're purging at every layer in the same spot in the middle, you're bound to create a big blob of plastic there that just hits the nozzle and becomes unstable at a certain point, with then a loose shell around it. For the second layer there will be a blob that rests on the brim inside the prime tower, the third layer will have a blob that rests on the blob of the second layer, the fourth layer will have a blob that rests on the blob of the third layer, and so on. So instead we spread out this blob and purge in a circular motion along the side of the prime tower. It's still using it as a purge bucket and some of these circles might not be supported from underneath, but you'll never get a blob of material building up. We also wipe the nozzle on the prime tower, if that setting is enabled, in a spot that evenly divides the blobs around the tower using the Golden Ratio.

> i dont know of you see what i mean, the problem is that if there is too much toolchange, the bucket will be full of material at some points .

Since it needs to purge a certain amount of volume, if the volume of the prime tower in total is not enough to contain this, it won't matter whether you purge in the middle or whether you purge along the side. The current implementation cuts off the priming if there is no more space for it, so the "minimum prime volume" won't actually be reached then. It should also give a warning value for the prime tower size then, I believe. The only solution then is to increase the prime tower diameter. I expect that the prime tower diameter is set by the profile author so that this never occurs.