Ultimaker/Cura #5290: [3.6.0] cura looks for image file in home dir, not /usr/share/cura/resources

CapnKernel, 2019-02-11:
**Application Version**
3.6.0

**Platform**
Linux, Fedora 28.  Self built, can recompile if you have suggestions

**Printer**
Ender 3 (but this problem isn't related to a particular printer)

**Steps to Reproduce**
Start cura in a terminal.

**Actual Results**
Stack trace (see below) while creating splash screen

**Expected results**
Splash screen

**Additional Information**

Screenshot showing non-redrawing window: https://imgur.com/a/ubANBcR

Strace shows that cura is trying to access `/home/mjd/.local/share/cura/3.6/images/cura-icon-32.png`:
```
stat("/home/mjd/.local/share/cura/3.6/images/cura-icon-32.png", 0x7ffe18d82fb0) = -1 ENOENT (No such file or directory)
```
It should be accessing `/usr/share/cura/resources/images/cura-icon-32.png`:
```
[mjd@xiaomao ~]$ ls -l /usr/share/cura/resources/images/cura-icon-32.png
-rw-r--r--. 1 root root 15000 Nov 13 01:47 /usr/share/cura/resources/images/cura-icon-32.png
```
If I cd into `/home/mjd/.local/share/cura/3.6` and create a symlink from `images` to `/usr/share/cura/resources/images`, then Cura is able to start.

I've looked at `UM/Resources.py`.  I think this should be looking in `/usr/share/cura/resources`, but I'm not sure of how to set `UM.Resources.__data_storage_path` to be `/usr/share/cura/resources`, or whether this is the right fix.  Perhaps there needs to be a change to Uranium, and perhaps I need to pass `/usr/share/cura/resources` to Uranium at build time.  Advice needed!

Stack trace:
```
[mjd@xiaomao ~]$ cura --debug
[MainThread] UM.Resources.__initializeStoragePaths [414]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [425]: Config storage path is /home/mjd/.config/cura/3.6
[MainThread] UM.Resources.__initializeStoragePaths [433]: Data storage path is /home/mjd/.local/share/cura/3.6
[MainThread] UM.Resources.__initializeStoragePaths [445]: Cache storage path is /home/mjd/.cache/cura/3.6
[MainThread] UM.Resources._copyLatestDirsIfPresent [463]: Found config: ['/home/mjd/.config/cura'] and data: ['/home/mjd/.local/share/cura']
[MainThread] UM.Resources._copyLatestDirsIfPresent [467]: Latest config path: /home/mjd/.config/cura/3.6 and latest data path: None
[MainThread] UM.Resources._copyLatestDirsIfPresent [475]: Same config path [/home/mjd/.config/cura/3.6], do nothing.
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [119]: Trying OpenGL context 2.0...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [128]: We got at least OpenGL context 2.0: 2.1 No profile
[MainThread] UM.Qt.QtApplication.initialize [138]: Detected most suitable OpenGL context version: 2.0 No profile
[MainThread] UM.Qt.QtApplication.initialize [144]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [148]: Initializing version upgrade manager ...
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_definition_changes] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_variant] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_material] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_quality] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_quality_changes] added.
[MainThread] UM.PackageManager._loadManagementData [76]: Bundled package management files could not be found!
[MainThread] UM.PackageManager._saveManagementData [117]: Package management file /home/mjd/.local/share/cura/3.6/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [92]: Loading plugin configuration file '/home/mjd/.config/cura/3.6/plugins.json'
2019-02-11 14:37:21,151 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ConsoleLogger
2019-02-11 14:37:21,182 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [80]: Found CuraEngine at: /bin/CuraEngine
2019-02-11 14:37:21,183 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraEngineBackend
2019-02-11 14:37:21,188 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalContainerProvider
2019-02-11 14:37:21,190 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJWriter
2019-02-11 14:37:21,200 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLReader
2019-02-11 14:37:21,202 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLWriter
2019-02-11 14:37:21,204 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJReader
2019-02-11 14:37:21,209 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimpleView
2019-02-11 14:37:21,209 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin ConsoleLogger was already loaded
2019-02-11 14:37:21,215 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UpdateChecker
[... 66 lines cut ...]
2019-02-11 14:37:21,729 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./plugins.json, skip it
2019-02-11 14:37:21,729 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./packages.json, skip it
2019-02-11 14:37:21,732 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./plugins.json, skip it
2019-02-11 14:37:21,732 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./packages.json, skip it
2019-02-11 14:37:21,862 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2019-02-11 14:37:21,864 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2019-02-11 14:37:21,865 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/bin/cura", line 138, in <module>
2019-02-11 14:37:21,865 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     app.run()
2019-02-11 14:37:21,865 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/usr/lib/python3.6/site-packages/cura/CuraApplication.py", line 686, in run
2019-02-11 14:37:21,866 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     super().run()
2019-02-11 14:37:21,866 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/usr/lib/python3.6/site-packages/UM/Qt/QtApplication.py", line 289, in run
2019-02-11 14:37:21,867 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     super().run()
2019-02-11 14:37:21,867 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/usr/lib/python3.6/site-packages/UM/Application.py", line 207, in run
2019-02-11 14:37:21,867 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self.startSplashWindowPhase()
2019-02-11 14:37:21,868 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/usr/lib/python3.6/site-packages/cura/CuraApplication.py", line 426, in startSplashWindowPhase
2019-02-11 14:37:21,868 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     super().startSplashWindowPhase()
2019-02-11 14:37:21,868 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/usr/lib/python3.6/site-packages/UM/Qt/QtApplication.py", line 228, in startSplashWindowPhase
2019-02-11 14:37:21,869 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self._tray_icon = QIcon(Resources.getPath(Resources.Images, self._tray_icon_name))
2019-02-11 14:37:21,869 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/usr/lib/python3.6/site-packages/UM/Resources.py", line 83, in getPath
2019-02-11 14:37:21,869 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     raise FileNotFoundError("Could not find resource {0} in {1}".format(args, resource_type))
2019-02-11 14:37:21,870 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: FileNotFoundError: Could not find resource ('cura-icon-32.png',) in 6
^C
[mjd@xiaomao ~]$ 

```

smartavionics, 2019-02-11:
Hello @CapnKernel , I run cura from a local installation happily enough using this script (cura.sh) which I have in /opt/cura/bin.

````
#!/bin/sh

# this assumes that $0 is equal to $CURA_HOME/bin/cura.sh after symlinks have been resolved

CURA_BIN=$(dirname $(realpath $0))

CURA_HOME=$(realpath $CURA_BIN/..)

CURA_LIB=$CURA_HOME/lib

PATH=$CURA_BIN:$PATH

export PYTHONPATH=$CURA_HOME/lib/python3/dist-packages:$CURA_LIB/cura
export LD_LIBRARY_PATH=$CURA_LIB
export QT_PLUGIN_PATH=$CURA_LIB/plugins
export QML2_IMPORT_PATH=$CURA_LIB/qml

cura $@
````

CapnKernel, 2019-02-11:
Hi @smartavionics.  Thanks for your message.  I can see that this will let you run Cura as a local install.  I'd like to run Cura as a system-installed package, ie, any logged in user can run it.  And for that to work, `/bin/cura` will have to look in `/usr/share/cura/resources`.  (This is what happens with Fedora's `cura` package)

smartavionics, 2019-02-11:
Ah, OK. Sorry, I can't really help with packaging.

Ghostkeeper, 2019-02-12:
The data storage path is being set to `$XDG_DATA_HOME` or `~/.local/share/` if that environment variable is missing. Maybe you can start it with that environment variable adjusted?

Do you know if we need to improve this in the source code to be more adherent to standards?

hroncok, 2019-02-12:
I suppose default icons are not what users should put it their homes, or are they?

Ghostkeeper, 2019-02-12:
Ah, no, sorry. I was referring to `Resources.__data_storage_path` which is indeed where it stores data files, not where it loads it from.

The places where it looks should be the list of:
* `$XDG_DATA_HOME/cura/4.0/resources`
* (if not running from source) `<executable>/../share/cura/resources`
* (if running from source) `../resources` relative to `CuraApplication.py` as well as `Application.py`

See also the function `CuraApplication.__addExpectedResourceDirsAndSearchPaths`.

hroncok, 2019-02-12:
In Fedora, we have resources in `/usr/share/cura/resources`, the icon for example lives in `/usr/share/cura/resources/images/cura-icon-32.png`.

`CuraApplication.py` resides in `/usr/lib/python3.7/site-packages/cura/CuraApplication.py`.

Should I patch something to make this work? I never heard about problems with this, until I saw this issue.

Ghostkeeper, 2019-02-12:
Because of the second place in my list, `<executable>/../share/cura/resources` I'd think it would work if your Python executable lives in `/usr/bin`. It would then look in `/usr/bin/../share/cura/resources`.

hroncok, 2019-02-12:
That'll be `/usr/bin/cura`, so yes, it should work.

CapnKernel, 2019-02-12:
@Ghostkeeper, I'll do some more digging.

CapnKernel, 2019-02-25:
I found the bug!  It's in Uranium, `UM/Application.py`'s `Application.getInstallPrefix()` method.

This bug will affect anyone who uses a Linux system where `/bin` is a symlink to `/usr/bin` (such as Fedora), _and_ the user has modified their `PATH` environment variable to have `/bin` before `/usr/bin`.

Within `getInstallPrefix()`, `sys.executable` is `/usr/bin/python3`, so `sys.argv[0]` is used.

In this case, running `cura` means `sys.argv[0]` will be `/bin/cura`, and `getInstallPrefix()` will return `/`.  `share` gets appended to this (ie, `/share/cura/resources`), and naturally, looking for resources here fails.

If `/bin` _isn't_ in the `PATH` before `/usr/bin`, running `cura` means `sys.argv[0]` will be `/usr/bin/cura`, and `getInstallPrefix()` will return `/usr`.  `share` gets appended to this (ie, `/usr/share/cura/resources`), and lookup of resources works.

To reproduce (on a machine where `/bin` is a symlink to `/usr/bin`, such as Fedora): 
```
/bin/cura
```
This gives the unresponsive window and backtrace, whereas this works:
```
/usr/bin/cura
```

One fix for the problem is to add `os.path.realpath()` to the chain in `getInstallPrefix()`, which causes the `/bin` symlink to be followed:
```
--- UM/Application.py.orig	2019-02-23 09:55:29.794764868 +1100
+++ UM/Application.py	2019-02-25 13:36:15.955003912 +1100
@@ -390,9 +390,10 @@
     @staticmethod
     def getInstallPrefix() -> str:
         if "python" in os.path.basename(sys.executable):
-            return os.path.abspath(os.path.join(os.path.dirname(sys.argv[0]), ".."))
+            exe = sys.argv[0]
         else:
-            return os.path.abspath(os.path.join(os.path.dirname(sys.executable), ".."))
+            exe = sys.executable
+        return os.path.abspath(os.path.join(os.path.dirname(os.path.realpath(exe)), ".."))
 
     __instance = None   # type: Application
 
```
While most people won't have `/bin` in their `PATH` before `/usr/bin`, it's valid to do so, and Cura should still work.  Please take my fix.

Ghostkeeper, 2019-02-25:
Fantastic analysis actually! Thank you for investigating.

hroncok, 2019-02-26:
Out of curiosity, @CapnKernel, why do you have `/bin` in PATH?

CapnKernel, 2019-02-27:
@hroncok, I added it to my `.bashrc` in 1993 on a non-Linux system (`/bin` and `/usr/bin` were separate)  and it's been there ever since! I've now taken it out (no longer needed eh!) but still, Cura shouldn't crash if invoked as `/bin/cura`.  

(Of historical interest: Original Unix systems had`/usr` on a separate volume, which was mounted relatively late in the boot process.  On such systems, all the binaries necessary to boot should be in `/bin`.  That doesn't apply to Linux)

I'm happy for this issue to be closed, esp since it's now covered in an issue under Uranium.  Thanks!

hroncok, 2019-02-27:
That's an old bashrc! Thanks for explaining.

closed 2019-03-08 by diegopradogesto

fieldOfView, 2019-03-09:
it's an older .bashrc sir, but it checks out