Ultimaker/Cura #5294: [3.6.0] Malfunctioning using "Lay Flat" when an object's tilted at 20 degrees

KeithPark, 2019-02-12:
**Application Version**
3.6.0

**Platform**
Windows 10 1809

**Printer**
Ender 3

**Steps to Reproduce**
1. Make a 65mm radius hex cylinder with bottom tilled at 20 degrees and filleted 2mm radius for all bottom corners.
(Make sure that the tilted bottom has flat surface)
2. Load the object in Cura 3.6.0
3. Rotate object by using "Lay Flat" function
4. Click Prepare

**Actual Results**
Watch the result each layer. The bottom layer will be printed partially. And it will grows wider in upper layer.

**Expected results**
The first layer should be aligned with the flat bottom surface.

**Additional Information**
[image]
[image]

[Cura_bug.zip](https://github.com/Ultimaker/Cura/files/2855702/Cura_bug.zip)
(This attached file includes a project file and a STL file)

Ghostkeeper, 2019-02-12:
It did lay your object flat. Just not on the biggest surface there is. Your model has rounded corners, meaning that there are lots of small surfaces in your curve that it could lay it flat on.

The Marketplace has a plug-in that does a bit better: The Auto-Orientation plug-in tries to find an orientation that minimises the need for support.

KeithPark, 2019-02-12:
> It did lay your object flat. Just not on the biggest surface there is. Your model has rounded corners, meaning that there are lots of small surfaces in your curve that it could lay it flat on. If it would choose the biggest surface there is, then it would've laid the container on its side which is also not what you wanted.
> 
> The Marketplace has a plug-in that does a bit better: The Auto-Orientation plug-in tries to find an orientation that minimises the need for support.

Thanks for the advise. I could fix the problem. But, I want Cura to be better.