Ultimaker/Cura #5296: Cura 3.6.0 Blank screen / no gui after startup

nicknugget, 2019-02-12:
**Application Version**
3.6.0 AppImage

**Platform**
XUbuntu 18.0.2 LTS amd64

**Printer**
Custom

**Steps to Reproduce**
Open / start Cura

**Actual Results**
 Blank screen. No usable GUI available. See attached screenshot.

**Expected results**
User interface showing.

**Additional Information**
Happened after a recent system update via update-manager. All my available cura versions down to version 3.5 show this behaviour. Version 3.4.1 and lower just work fine.

[image]

Ghostkeeper, 2019-02-12:
Not seeing this on normal Ubuntu 18.10 right now. Do you know what packages were updated? It appears like something broke in the Qt library.

nicknugget, 2019-02-12:
Unfortunately I can't remember which packages were updated during the update. I didn't pay any attention to this at first. 

On another computer with a current Xubuntu 18.10 it doesn't work for me either.

I also suspect that it has something to do with Qt. Unfortunately, I don't know enough about it.

nicknugget, 2019-02-12:
I added the stderr.log from cura. It contains all messages from the last program start.

[stderr.log](https://github.com/Ultimaker/Cura/files/2857529/stderr.log)

Ghostkeeper, 2019-02-13:
That log looks fairly normal. Nothing special to see that could be related to the GUI missing.

mboberlin, 2019-02-13:
Same problem here.. also Ubuntu 18.04.2 LTS.

nallath, 2019-02-13:
But it should just use the qt that we ship with the appimage right? So that pretty much leaves graphics drivers as the culprit.

mboberlin, 2019-02-13:
Hi,
yes, i am using the original appimages. Problem exists in Cura 3.5 to
4.0beta.
Greetings,
Michael

nicknugget, 2019-02-13:
At least it looks like i'm not alone with the problem. The funny thing is, that it works just fine in a fresh XUbuntu 18.10 installation with all current system updates running in a virtual machine (virtualbox)

mikeyancey, 2019-02-13:
I'm also having this issue (Cura 3.6.0, Ubuntu 18.04 LTS).
(For me, anyway) the menus are 'there', if you click about - but they're blank. 
Right-click menu is present, but blank.
I have variously deleted .cache, .config, and .local/share/cura items, with no effect. 
I'll attach cura.log (from .local/share/cura/3.6/). There are warnings: "UM.Qt.QtApplication.__onQmlWarning"...

[cura.log](https://github.com/Ultimaker/Cura/files/2862314/cura.log)

mboberlin, 2019-02-13:
I have a Thinkpad T510 with Intel Ironlake Mobile Graphics.
It really seems to be a problem with the graphics driver, using a more actual driver from the PPA mentioned in [https://askubuntu.com/questions/1065852/how-to-upgrade-intel-graphics-driver]  it works again.

mikeyancey, 2019-02-13:
Aha - I was about to wipe all traces of Cura's files and re-install.
In ~.cache folder, there's a cache of 'qtshadercache'.
I saved it (mv qtshadercache to xyzzy-qtshadercache - shouldn't matter if I deleted it, it should re-gen, but - saving it in case it's important).
RUN (for the 45th time today) - Cura 3.6.0
ALL menus are back; right-click menu back; the Printer side-bar is back.
[image]

I believe in my instance, Cura was shutdown (forced) improperly; it wasn't a 'QT was updated' or 'Display driver was updated' issue.
So, possibly QT cache related - it's clearly going to the qtshadercache (under .cache), but receiving some 'half-baked' info.

mikeyancey, 2019-02-14:
Glad you got it working. Very cool.

Well - I guess the problem can be arrived at both ways, then.
A driver that causes complaint from QT and a corrupt 'QT Shader Cache' have
the same effect.

I swear, mine looked exactly as the original screencap - white panel where
the 'Selected Printer' should be... etc.
..
Mike Yancey
Dallas, Texas
https://www.mikeyancey.com <http://www.mikeyancey.com>

nicknugget, 2019-02-14:
Previously i tried to get rid of all cura-folders in ~./ - nothing changed. 

As suggest by @mikeyancey I deleted the ~./cache/qtshadercache folder this morning. 

YES! Cura is working again!

wgaonar, 2019-02-19:
So many thanks to @mikeyancey. It is working now

kitts82, 2019-02-24:
> Aha - I was about to wipe all traces of Cura's files and re-install.
> In ~/.cache folder, there's a cache called: 'qtshadercache'.
> I saved it (mv qtshadercache to xyzzy-qtshadercache - shouldn't matter if I deleted it, it should re-gen, but - saving it in case it's important).
> RUN (for the 45th time today) - Cura 3.6.0
> ALL menus are back; right-click menu back; the Printer side-bar is back.
> [image]
> 
> I believe in my instance, Cura was shutdown (forced) improperly; it wasn't a 'QT was updated' or 'Display driver was updated' issue.
> So, possibly QT cache related - it's clearly going to the qtshadercache (under .cache), but receiving some 'half-baked' info.

Thanks. This fixed it for me.

Ghostkeeper, 2019-02-25:
Cura is not the only application whose Qt shaders get cached in that folder. The files in that folder are not labelled to belong to which application. So it's hard for us to reset that folder...

dvoros, 2019-11-03:
Faced the same issue after the update of the following qt-related packages (standard update with 18.04):
```
qt5-gtk-platformtheme:amd64 (5.9.5+dfsg-0ubuntu2.1, 5.9.5+dfsg-0ubuntu2.3)
libqt5dbus5:amd64 (5.9.5+dfsg-0ubuntu2.1, 5.9.5+dfsg-0ubuntu2.3)
libqt5widgets5:amd64 (5.9.5+dfsg-0ubuntu2.1, 5.9.5+dfsg-0ubuntu2.3)
libqt5printsupport5:amd64 (5.9.5+dfsg-0ubuntu2.1, 5.9.5+dfsg-0ubuntu2.3)
libqt5gui5:amd64 (5.9.5+dfsg-0ubuntu2.1, 5.9.5+dfsg-0ubuntu2.3)
libqt5core5a:amd64 (5.9.5+dfsg-0ubuntu2.1, 5.9.5+dfsg-0ubuntu2.3)
libqt5network5:amd64 (5.9.5+dfsg-0ubuntu2.1, 5.9.5+dfsg-0ubuntu2.3)
```
Removing `~/.cache/qtshadercache/`, as suggested by @mikeyancey, resolved the issue. Thanks for that!

Ghostkeeper, 2019-11-04:
If the issue is related to upgrading Qt version, then maybe it would also happen when using multiple applications with different Qt versions. Cura's AppImage always uses Qt 5.10 (for the latest couple of Cura versions anyway). So if a different application uses different Qt versions it might get corrupt?

Ghostkeeper, 2020-02-12:
In Cura 4.5 we've added a workaround for this issue. These are the steps to apply this workaround:

* Make sure that Cura is not running.
* Go to the configuration storage location for Cura. On Windows that's `C:\Users\<username>\AppData\Roaming\cura\4.5`. On MacOS that's `~/Library/Application Support/cura/4.5`. On Linux that's `~/.config/cura/4.5`.
* Open the cura.cfg file with a text editor.
* Under the `[view]` header, add a new key:
      `force_empty_shader_cache = True`

This will cause Cura to not use the shader cache any more and clear its part of the cache. It'll fix your issue in the same way as removing the `qtshadercache` folder every time you start Cura. The flipside of this is that your start-up will take a performance hit of 1 or 2 seconds. Given how rare this is, we don't like to disable the shader cache by default. And given that it tends to get corrupted sporadically (not repeatedly on the same computer), it might just be easier to find and delete that `qtshadercache` folder just once.

We're still looking for a better way to fix this, but we have no idea how. Ideally we'd be able to detect somehow that things got corrupted.

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

nicknugget, 2024-11-15:
I have not encountered this problem in the newer versions of Cura (5.8.0 and up) since then.

GregValiant, 2024-11-15:
Thanks.  I go ahead and close this.

closed 2024-11-15 by GregValiant