Ultimaker/Cura #5297: [4.0 Beta] Tree supports significantly harder to remove

CHINCB, 2019-02-12:
**Application Version**
4.0 Beta/3.6

**Platform**
Windows 10

**Printer**
Wanhao D6

**Steps to Reproduce**
Slice model with tree supports enabled on 4.0 Beta

**Actual Results**
Tree supports are significantly more difficult to remove when using 4.0 to slice. I use cura to slice tabletop miniatures which can be quite delicate and when using exactly the same settings the supports from 4.0 are much stiffer/harder and tend to tare the print.

**Expected results**
Same as 3.6

**Additional Information**
Printed at layer height of 0.08mm

CHINCB, 2019-02-13:
Can I provide any additional information?

Ghostkeeper, 2019-02-14:
I'm not seeing any difference when I slice a simple overhang model: [UM3_basic_overhang.zip](https://github.com/Ultimaker/Cura/files/2864365/UM3_basic_overhang.zip). In both 3.6 and 4.0, the model starts at layer 398 and the support ends at layer 394. The support distance is still too high in both versions (which is a known bug but not yet solved).

Maybe you can provide a project file that shows the problem?

CHINCB, 2019-02-15:
When reviewing the preview for each of the models the only difference I see is the number of walls for the tree support (It seems like it is always 2 in 4.0). The difference becomes really pronounced after printing. What I will do is slice the suggested overhang file in both versions, upload the gcode but also print both and see if i can get images of the difference. Will revert back with info during the next 24 hours.

Ghostkeeper, 2019-02-18:
Then I'd take a very close look at the Tree Support Wall Thickness and the Tree Support Wall Line Count settings in both versions.

tthfry, 2019-04-26:
tl;dr: Make sure you set your Support Density to calculated for your miniature printing profile!

I am having the exact same problem -- printing minis and the tree support behavior changed. After investigating a couple of older slices and a current slice, it appeared to me that the tree supports started to respect the support pattern of the regular support. This resulted in double walls for the tree support as well as additional structure between the walls.

After investigating some more and finding out that the default presets do not suffer from this (and it also happened to me in 3.6), I noticed that the Support Density had an explicit value set in my profile. If I reset the Support Density to calculated, activating Tree Support sets that value to 0% and deactivating sets it back to 15%.

So the issue here seems to be that Tree Support can't handle the explicitly set regular support density setting.

closed 2019-04-29 by Ghostkeeper

Ghostkeeper, 2019-04-29:
Indeed some profiles overwrite the default formula with a fixed value, which would set tree support to have that same fixed density as well. I've fixed that up now. Hopefully that resolves this issue.

Only for PVA I've left the density to be fixed (the same as normal support).

commit 25bbc1e6 (2019-04-29): Set default infill density for tree support to 0% in profiles

In profiles that overwrite the support infill density, also keep it 0% for tree support.
Except PVA profiles. Those are probably better off with their normal infill density.

Fixes #5297.