Ultimaker/Cura #5299: Does not abort

nkuck, 2019-02-12:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

nallath, 2019-02-13:
Please fill in the entire template. This is not nearly enough information for us to know what the actual problem is.

Ghostkeeper, 2019-02-13:
In my experience, Github does abort quite easily. Look, I can demonstrate it here with this issue.

(Please fill in the issue template. It puts a warning there that we close your issue if it's not filled in.)

fieldOfView, 2019-02-13:
When you fill in, make sure you remove the `` bits. Don't put your text in between them, because that will make the text invisible.

Ghostkeeper, 2019-02-13:
In the comment markup of the message there was indeed a properly filled out template! I removed the comment markup so that it's actually visible to people who can't edit stuff...

So I'm assuming that this has to do with USB printing. I can't reproduce this though. I don't have your printer, but I tried with an Ultimaker 2+ using these steps:
* Start Cura and add an Ultimaker 2+ printer.
* Connect a printer with a USB cable.
* Load a cube.
* Start printing via USB. The printer starts moving immediately because it's UM2+ and USB printing doesn't properly heat up.
* After 10 seconds, press abort. The printer homes and then stops moving. All seems to work fine.

nkuck, 2019-02-13:
Thanks for the feedback on my issue presentation. Here is/are the issue(s):
With Cura started, and connected via USB to my Prusa i3 clone, it is sometimes recognized, sometimes not (issue #1). My computer is a MacBook Pro running latest iOS OS Mohave V 10.14.4 Beta (18E184e).
Issue #2:
Normally after I click on Prepare, I just get a Save to File option, and not Print over USB, as I mentioned above, the computer/Cura do not always talk to each other.
When/if I get a Print option, it usually follows that I will get an Abort option which, when selected, nothing happens. The print operation will just continue.
Running Marlin FW Bugfix 1.1.x

Ghostkeeper, 2019-02-14:
Maybe you could share a log? Maybe that shows something?

nkuck, 2019-02-14:
Sure. Which log would be useful?

Live simply!

Ghostkeeper, 2019-02-14:
Cura's GUI log file. See https://github.com/Ultimaker/Cura#logging-issues

nkuck, 2019-02-14:
I tried it this morning and all seems fine. Not sure why.
Thanks for your assistance and interest.

nkuck, 2019-02-14:
Tried it again but failed. Files are attached. Thanks for the review.

Ghostkeeper, 2019-02-18:
Attachments to the e-mail are not published via Github. Please go to the Github website and upload it there: https://github.com/Ultimaker/Cura/issues/5299