Ultimaker/Cura #5302: Ironing each layer.

LazaroFilm, 2019-02-13:
Please add a option to iron each layer in a print at 100% infill. 
This option could allow to have significantly more transparent print with transparent filaments as pointed out in this post: https://community.ultimaker.com/topic/26593-please-include-an-iron-every-layer-option-for-clearer-prints/

nubnubbud, 2019-03-07:
> Please add a option to iron each layer in a print at 100% infill.
> This option could allow to have significantly more transparent print with transparent filaments as pointed out in this post: https://community.ultimaker.com/topic/26593-please-include-an-iron-every-layer-option-for-clearer-prints/

Original poster here: as well, the process seems to produce stronger parts by remelting and reinforcing the part at a crystalline level. I have studied the parts produced with a microscope to confirm.

Ghostkeeper, 2019-03-07:
This is currently in our backlog, meaning it is our wish to implement it but it's unclear of when we can do that. At least something similar, to iron all skin layers instead of just the top one. It's easy to implement in TopSurface.cpp (just remove 3 lines that remove part of a layer where something else is in the layer above it) but would produce a bit of ugly architecture if done like that.

As a hobby project aside of work, I removed that check and printed a lens with it. But that was no success, as it produced a gigantic amount of overextrusion which introduced all sorts of blobs and air pockets and eventually ripped the print from the build plate. I then printed it without ironing and that was more of a success.

nubnubbud, 2019-03-07:
That is very odd, considering all my attempts were simply inserting the same sampled lines of ironing code after every layer. It only works for "tower" style objects, but produced them flawlessly to within a tenth of a millimeter. It was printed without changing any other settings, because this is a method specifically to be able to avoid that blobbing and overextrusion. Because cura's code is a lot more friendly than other slicers, I may simply be able to make a plugin by studying it, given a week or two.

megablue, 2019-03-07:
i made the same suggestion #3065 (Ironing on every or/and N layers) awhile ago, but my original intend was just filling the voids for stronger part despite knowing that it will make a transparent part looks better.

Ghostkeeper, 2019-03-08:
Ah, yeah thanks Megablue. Let's close this one as a duplicate then.

My attempt failed but to print transparently I had set the temperature quite high and the layer height quite low, so maybe it just doesn't work well combined with those things. It's not really a reason to not do this. Just one print, you know.

closed 2019-03-08 by Ghostkeeper