Ultimaker/Cura #5303: [4.0.0-beta] Does not retract on travel even if longer than specified distance

frank26080115, 2019-02-13:
**Application Version**
4.0.0-beta

**Platform**
Windows 10 Pro 64 bit

**Printer**
Ultimaker 2 using a 0.8mm nozzle

**Steps to Reproduce**
Object is something that I need a very nice looking bottom surface

- Enable retraction
- "Retraction Minimum Travel" set to something really short like 1.6mm (double nozzle size)
- "Max Comb Distance with No Retraction" set to 0
- "Retract Before Outer Wall" enabled
- "Combing Mode" set to "All"

Slice

**Actual Results**
There are still travels well over 1mm that does not start with a retract and end with an unretract, ruining the look of the print

Examining the gcode file, yes there are G1 instructions that happen after a G0 instruction without a G11 instruction, in the first layer, with type WALL-INNER or WALL-OUTER

line 167 of the gcode example (see attached)
```
G1 X17.445 Y161.372 E138.5655
G1 X17.801 Y161.436 E138.65231
G0 X17.784 Y161.533
G0 X43.366 Y176.373
G0 X43.427 Y176.295
G1 X43.647 Y176.463 E138.71874
G1 X43.77 Y176.557 E138.7559
```

As you can see, that move is clearly hug, a 26 mm long move, when Retraction Minimum Travel is set to only 1.6mm

The result looks ugly because the start of some lines are tapered and doesn't stick to the bed as well as a move that starts off with a full unretract. This leaves a small divot in the print.

**Expected results**
No travel moves should happen that does not start and end with retract and unretract respectively.

the example gcode above should look like
```
G1 X17.445 Y161.372 E138.5655
G1 X17.801 Y161.436 E138.65231
G0 X17.784 Y161.533
G10
G0 X43.366 Y176.373
G0 X43.427 Y176.295
G11
G1 X43.647 Y176.463 E138.71874
G1 X43.77 Y176.557 E138.7559
```

**Additional Information**

[bugreport_feb13.zip](https://github.com/Ultimaker/Cura/files/2862604/bugreport_feb13.zip)

Attached project, also the bad gcode, and the corrected gcode (I corrected it with a Python plugin I wrote just for this problem)

frank26080115, 2019-02-13:
[image]

here's an image of the difference between the gcode with the bug, and the gcode after I manually corrected it

frank26080115, 2019-02-14:
here's the plugin I wrote, it's my first plugin, so take it with a grain of salt

[FirstLayerAlwaysRetract.zip](https://github.com/Ultimaker/Cura/files/2862749/FirstLayerAlwaysRetract.zip)

smartavionics, 2019-02-14:
You need to set combing mode to "no skin" and max comb distance with no travel to something like 10 (i.e. small but > 0).

Ghostkeeper, 2019-02-14:
"Retraction Minimum Travel" makes sure that there are no retractions for travel moves that are smaller than this minimum distance. It doesn't force retractions for travel moves that are larger than this minimum distance. That would be really bad too because there are more conditions required for retractions. Retractions:

* they take time, which also causes a blip on the surface,
* they grind the filament at the feeder, and
* they cause a flow disruption.

For instance there is the setting "Maximum Retraction Count" combined with "Minimum Extrusion Distance Window" which attempts to prevent the filament from grinding and/or the flow disruption by limiting the number of retractions in a certain window of extruded filament.

To limit the effect of retractions, Cura doesn't normally retract during travel moves that only go through skin, support or infill, but you can customise that for skin with the Combing Mode setting and for support with the Limit Support Retractions setting.

closed 2019-02-14 by Ghostkeeper