Ultimaker/Cura #5307: [4.X GIT] Slice Summary Usage Garbled

Enverex, 2019-02-14:
**Application Version**
GIT Master (4.x)

**Platform**
Linux x86_64

**Printer**
Monoprice Ultimate / Duplicator 6

**Steps to Reproduce**
Add any model, slice, hover over the (i) icon next to the sliced model information.

**Actual Results**
Text appears stacked and garbled in the corner.

**Expected results**
Not that.

**Additional Information**
Probably easier to just see the issue rather than trying to explain it.
[image]

Not sure if you're just in the middle of redesigning this and it's not really a bug, but thought I should report it, just in case.

nallath, 2019-02-14:
Hmm, strange. I've just checked master on my machine and there it does look correct.

Ghostkeeper, 2019-02-18:
Also can't reproduce it here. Maybe it was just temporary?

Enverex, 2019-02-18:
Just recompiled from source, still happening. Could it be a font/DPI/desktop environment issue?

[image]

Ghostkeeper, 2019-02-19:
Or maybe Qt version? Cura's builds are using Qt 5.9. I'm running Qt 5.11 when running from source.

Enverex, 2019-02-19:
Apparently I'm running Qt 5.12.1 on that machine.

Ghostkeeper, 2019-02-19:
Well, if you manage to fix it for 5.12, we'd love your fix :) Eventually we'll upgrade to more modern Qt versions once we can drop support for MacOS 10.11.

Enverex, 2019-02-19:
I can live with it being broken as it's not something I really use. I just thought it was worth reporting given that I wasn't expecting it to be designed to look like that ;)

wilensky, 2019-03-28:
Confirming on _ArchLinux_ + CURA `4.0.0` + Gnome 3.32.0.
The balloon with time and material info is opened garbled in both fullscreen and windowed modes.

> [image]

fieldOfView, 2019-03-28:
Use the AppImage.
The problem is the version of Qt. The AppImage includes Qt 5.10, your ArchLinux install probably uses Qt 5.12

wilensky, 2019-03-28:
@fieldOfView, yes. Qt is 5.12.2

```
Qt 5.12.2 (x86_64-little_endian-lp64 shared (dynamic) release build; by GCC 8.2.1 20181127) on "xcb" 
OS: Antergos Linux [linux version 5.0.4-arch1-1-ARCH]
```

Enverex, 2019-03-28:
@wilensky For future reference, I wouldn't call Antergos "Arch" as they're still different distros with different setups.

wilensky, 2019-03-28:
@Enverex, indeed. My bad. Working on several machines and didn't pay attention from which I reported.

closed 2019-03-29 by nallath

Enverex, 2019-03-29:
As this is closed, is there a bug open for QT 5.12 compatibility?

Ghostkeeper, 2019-03-29:
For the moment we don't want to switch our builds to Qt 5.12, since that would drop support for older MacOS versions that our users are still using.

That said, since this week I'm also on Qt 5.12 and have this issue as well so it's a matter of how long I can live with this issue I guess :wink:

wilensky, 2019-03-29:
[`v4.0.0` AppImage](https://github.com/Ultimaker/Cura/releases/download/4.0.0/Cura-4.0.0.AppImage) works like a charm. Reported issue not reproducing also no fonts problem.
Thanks!

Enverex, 2019-04-15:
I think this may now be affecting the machine settings page, as it now looks like this and is essentially unusable:
[image]

Sometimes these fields are also uneditable, for unknown reasons. Looks like the AppImage may now be the only usable option on Qt 5.12 systems.

Ghostkeeper, 2019-04-15:
Yeah we saw that too. It's introduced by recent changes to this page to make it work in our new onboarding workflow. We'll make sure it's readable again before launch.