Ultimaker/Cura #5308: [3.6.0] Printing walls of odd multiples of nozzle size

theWaldschrat, 2019-02-14:
**Application Version**
3.6.0 (no other version tested)

**Platform**
Windows (but should be independent)

**Printer**
Creality Ender 3
[Hollow-Wall-1200mu.zip.zip](https://github.com/Ultimaker/Cura/files/2866292/Hollow-Wall-1200mu.zip.zip)

**Steps to Reproduce**
1. Default
  - Open a model with wall that are a precise multitude of the nozzle size, e.g. 0.4, 1.2 or 2.0 mm. A simple hollow square shaft is sufficient.
  - Select a default profile, e.g. draft.
  - Have it sliced.
  - (Up to here can be seen in attached Hollow-Wall-1200mu-Default.3mf)
2. Horizontal Expansion
  - Reconfigure your nozzle size to 0.399 mm or configure "Horizontal Expansion" to 0.001 mm.
  - For 0.4 and 1.2 mm setting "Outer Wall Inset" to -0.001 also helps, but not for 2 mm.
  - Setting line width to 0.399 helps doesn't help for 0.4 mm, but for the thicker walls.
  - Slice again.
  - (Up to here is in attached Hollow-Wall-1200mu-Expanded.3mf)
3. Minimal Wall Flow
  - Set "Minimum Wall Flow" to 1%.
  - Slice again
  - (Now we're in Hollow-Wall-1200mu-Expanded-MinFlow.3mf)

**Actual Results**
1. Default
  - 0.4 mm wall: Not sliced at all.
  - 1.2 / 2 mm wall: Sliced, but with terrible wiggly fill between lines which is extremely slow to print and will shake your printer apart.
2. Horizontal Expansion
  - All wall thicknesses are as expected printed from parallel lines.
  - But the innermost line (which is also the only line for 0.4 mm) is actually printed "twice", but the second pass is with minuscule extrusion of about 0.001 mm filament (About 1 E-step).
3. Minimal Wall Flow
  -  Now the model is printed as expected with the walls made from parallel lines.

**Expected results**
See results obtained with "Horizontal Expansion" and "Minimal Wall Flow".

**Additional Information**
Nearly all printers in the world print (by default) with 0.4 mm nozzles, so most models are designed with wall that are a multiple of 0.4. Particularly 1.2 and 2 mm can be found very often. Because people expect walls like that to be printed nice and quick and solid with parallel lines.

theWaldschrat, 2019-02-14:
Ah, forgot to mention: Walls with even multiples of nozzle size, e.g. 0.8 and 1.6 mm are printed just fine with the default settings.

Ghostkeeper, 2019-02-18:
Actual result 1a is expected behaviour. It deems this too small to print because after an inset to determine the centerline of your wall, it gets a perimeter of 0 length which is then left out.

Actual result 1b should be fixed in Cura 4.0.

Actual result 2b is expected. All walls that Cura generates are loops. This is a limit of our implementation which makes a whole lot of things simpler. Luckily the Minimum Wall Flow is meant to prevent that, but we haven't tested that enough yet to make it the default.

closed 2019-02-18 by Ghostkeeper

theWaldschrat, 2019-02-23:
I wouldn't necessarily say that "expected by the devs" is the same as "expected by the user". At least I did not except it. But anyway, nice that at least the weird infill will be fixed.
Also I'd be interested to have a look at the algorithms myself, would someone be willing to point me to where the paths are calculated (and in some cases dropped). I guess it ought to be somewhere in the CuraEngine repository.

Ghostkeeper, 2019-02-27:
Yeah, but in this case it is not only "expected by the devs" but also "accepted as a shortcoming" because we see no viable alternative implementation.

The method that generates the wall centrelines is here: https://github.com/Ultimaker/CuraEngine/blob/cf1a3515fe618e0cd5b90378868840a2dda97503/src/WallsComputation.cpp#L24