Ultimaker/Cura #5309: UM3 - Adjust printing speed via menu @UM3 for extruder 1 is also changing the speed for extruder 2

ruhnselb, 2019-02-14:
Using Ultimaker UM3 - Firmware 4.3.3.20180529.

While printing I want to change the printing speed for material 1 via printers menu. The speed is changing immediately but when the printer is switching to the second Extruder (support material) I notice that also the printing speed for material 2 was changed.

Is this the intended behaviour or a bug? Does anybody have an idea how to overcome this problem?

ChrisTerBeke, 2019-02-14:
This repo is for Cura, not UM3 firmware. Some firmware devs might look at this repo, but it's not likely. The best route to report this bug would be via your reseller.

closed 2019-02-15 by nallath

Ghostkeeper, 2019-02-18:
(This is intended behaviour of the printer. The implementation is too limited to track which speed to adjust to for which extruder because the display doesn't know which extruder is currently actively printing.)