Ultimaker/Cura #5310: [3.6.0] Support blocker resize wrong direction

Festivejelly, 2019-02-14:
**Application Version**
3.6.0 + (havnt checked older versions)

**Platform**
Windows 10

**Printer**
N/A

**Steps to Reproduce**
Add a support blocker mesh, then go to resize, uncheck uniform resizing and drag the Y hand to scale it on the Y, it instead scales on the Z axis.

nallath, 2019-02-15:
I've just tried this on master (which is a fair bit ahead of 3.6) and I've not been able to reproduce this. Could you make a video of the bug in action?

Ghostkeeper, 2019-02-18:
Maybe it happens only after the model itself that the block was linked to was rotated 90 degrees?

Ghostkeeper, 2019-02-18:
Or it could be another duplicate of this bug report: https://github.com/Ultimaker/Cura/issues/4136

Festivejelly, 2019-09-19:
Can confirm this is a duplicate of the referenced bug report.

closed 2019-09-19 by Festivejelly