Ultimaker/Cura #5311: [4.0.0 Beta] Extruder Start/End G-code not written to output G-code

beckbw, 2019-02-15:
**Application Version**

4.0.0 Beta

**Platform**

Mac OS mojave 10.14.3

**Printer**

Creality CR-X. (Dual Extruder) Definition and Dual Extruder JSON(s) from this RAR from Creality:
https://www.thingiverse.com/asset:124070

**Steps to Reproduce**

- Dual Extrusion Print: TRUE
- Slice
- Examine G-code.
- Neither Extruder Start G-code or Extruder End G-code (from Machine Settings) written when filament swapped
[image]
**Actual G-code: Transition to T0:**
G92 E0
T0
G92 E0
**Actual G-code: Transition to T1:**
G92 E0
T1
G92 E0

**WORK AROUND**: Use TWO search-n-replace REGEXP post-processing scripts (one each for T0,T1):
e.g.
**ORIG**: G92 E0\nT1\nG92 E0
**NEW**: G92 E0\nG1 F2400 X15 Y15 E-80 ;Retract 80mm while homing\nG92 E0\nT1\nG92 E0\nG1 F2400 E80 ;Feed 80mm\nG92 E0
[image]
**Actual G-code with scripts: Transition to T1:**
G92 E0
G1 F2400 X15 Y15 E-80 ;Retract 80mm while homing
G92 E0
T1
G92 E0
G1 F2400 E80 ;Feed 80mm
G92 E0

 Extrus

**Actual Results**

see above
**Expected results**

Extruder G-code in Machine Settings should insert during tool swap.

**Additional Information**

Ghostkeeper, 2019-02-18:
Good bug report. Nice and detailed, and you even put a workaround there.

I'm not reproducing your issue but got something else buggy. There seems to be an encoding mistake in the extruder start g-code, but not in the extruder end g-code. I put comments in the start and end g-code of both extruders, and then got the following file: [test.zip](https://github.com/Ultimaker/Cura/files/2875594/test.zip)

This includes snippets like:

```
G1 F1200 E19.20313
G92 E0
;EXTRUDER1 END
T1
G92 E0
0;�ogU
M109 S205
M104 T0 S160.8
G1 F1500 E-6.5
```

In place of that weird `ogU` thing it should've said `;EXTRUDER2 START`.

beckbw, 2019-02-18:
The test gcode you sent has A  LOT of weird control characters. e.g.:
*line 5680:*
G92 E0
;EXTRUDER2 END
T0
G92 E0
�UX
M109 S190

Is that using the Creality JSONs?

None of my gcode has control characters like that near the tool transitions (I just checked).

Ghostkeeper, 2019-02-19:
No, this was done by modifying the Ultimaker 3 printer in the machine settings. I set `;EXTRUDER1 START`, `;EXTRUDER1 END`, etc. in the start/end g-codes for extruders.

beckbw, 2019-02-19:
Tested removing " ; Comments" from machine settings extrude gcode and that had no effect on lack of insertion.

Ghostkeeper, 2019-03-15:
Found the bug: https://github.com/Ultimaker/CuraEngine/commit/8abc59d60e8b6fe3e0a71f714786c5f78c1666bb

closed 2019-03-15 by Ghostkeeper

geekhed, 2019-03-17:
Awesome you found the problem and fixed it!  I have been having this problem for months.  But--how do I get this fix into my Cura 3.6.0?  I'm not a programmer so need your direction.

Ghostkeeper, 2019-03-18:
Well, you don't, or hardly. It requires quite some programming experience.

The way to bring this fix into 3.6 would be to clone the source code of https://github.com/Ultimaker/CuraEngine, checkout the 3.6 branch. Then run this command in a console: `git cherry-pick 8abc59d60e8b6fe3e0a71f714786c5f78c1666bb`, resolving any conflicts that might arise. And then build CuraEngine from your new source. You then get an executable CuraEngine that you could paste inside your Cura installation.

The building CuraEngine from source though is kind of involved and arduous and I wouldn't wish that upon any non-programmer. I'm afraid you'd have to wait until Cura 4.1.

In the meanwhile, you can instead use the Search and Replace post-processing script to place correct g-code around every extruder switch. You could search for the string `T0` and replace it with `;Extruder 2's end g-code\nT0\n;Extruder 1's start g-code` and similarly for the string `T1`. That's a workaround. If you're having trouble with the workaround, I could give more step-by-step instructions if you can tell me which printer you're using.

Block137, 2019-03-23:
I still get this problem with 4.0.0 release.
Installer was downloaded 21st Mar.

Ghostkeeper, 2019-03-26:
Yes, it was fixed for the 4.1 release. The 4.0 release was on its way out by the time we got around to fix it. Sorry!

PaulF8080, 2019-04-08:
I'm not getting the start/end code. Does that mean I need 4.1. The reason I'm asking is the discussion added other bugs so I am not clear if the bug in the topic title is fixed in 4.0 or 4.1. Where can I get 4.1. , if needed?

Ghostkeeper, 2019-04-12:
Yeah the bug in the topic title should be fixed in 4.1. I'm not able to reproduce it any more now and we have some tests set up to verify it.

PaulF8080, 2019-04-13:
I have a CR-X found out I don't need extra gcode so erasing the start end code and the funny characters disappeared.  I looked at the characters and they one or three bytes long each filament switch.

Printerman5, 2019-06-13:
I got Cura 4.1 but the g-code for it isn't working. I got a JGaurora a5 3d printer but none of the prints I have tried even started. Any help would be greatly appreciated.