Ultimaker/Cura #5313: [3.6.0] Filament change retraction is backwards

Kinlateral, 2019-02-15:
**Application Version**

3.6.0

**Platform**

Windows 10

**Printer**

Ender 3

**Steps to Reproduce**

Enable filament change in post processing

**Results**

When enabling "Filament Change" in post processing, it defaults to an Initial Retraction value of 30.0mm and a Later Retraction Distance of 300.0mm. However, these values actually result in extrusion rather than retraction. The values need to be set to negative to make it actually retract.

Ghostkeeper, 2019-02-18:
That seems to be printer-dependent then, because we originally had it backwards and then got the same bug report. The filament change is implemented with the M-code `M600`, so it could be that different firmwares are interpreting it differently. The [RepRap Wiki](https://reprap.org/wiki/G-code#M600:_Filament_change_pause) seems to say that it should be a positive number, because it says "Retract Distance" rather than something like "Delta E".

fiddlekrazy, 2019-02-22:
I the Ender 3 and had the same problem. 'Ghostkeeper is right, you have to make it a negative to work.

IvenBach, 2020-11-22:
Application Version: 4.6.1
Platfrorm: Windows 10
Printe: Ender 3 Pro

Just bit by this issue.
Here to confirm the `Initial Retraction` is backwards and extrudes filament. Submitting a value of `50` and saving the gcode emits `E50.00`. Changing it to `-50` does **NOT** emit any command.

GregValiant, 2023-09-14:
Can this be closed or is it still an issue?

closed 2023-09-29 by github-actions[bot]