Ultimaker/Cura #5315: [Cura 4.x] Tree Supports Only Built Within Small Base Area

Enverex, 2019-02-15:
**Application Version**
GIT Master (4.x)

**Platform**
Linux x86_64

**Printer**
Monoprice Ultimate / Duplicator 6

**Steps to Reproduce**
Add a large model that has big overhanging areas outside of the main mass. Enable tree supports.

**Actual Results**
Tree supports appear to only start from within the (rather small) skirt area on the base rather than being positioned whenever they're needed.

**Expected results**
Tree supports should start from wherever they're needed on the baseplate, which is the same behaviour as normal supports.

**Additional Information**
Here's a model with a large overhang that needs support. Note the lack of supports on the wings and that no supports appear outside of the zone within the skirt:
[image]

Increasing the allowed tree support angle and making the model a little smaller results in a bit of support being created, but only at an extreme angle from the base tree, rather than creating more trees on the build-plate to support areas that need support:
[image]

Turning on normal supports works as you'd expect. It builds them wherever it needs them on the base plate:
[image]

I can't find any option that explains why tree supports aren't being created in more sane locations.

Ghostkeeper, 2019-02-18:
If you use normal support generation but then disable the "Use Towers" option, does it then similarly not create ordinary support?

Ghostkeeper, 2019-02-18:
I found your dragon on Thingiverse, but for me it does create proper support under the wings.

[image]

Maybe this has been fixed in 4.0 then? Or maybe it's just because I'm using different settings?

Ghostkeeper, 2019-02-18:
Looking at your second screenshot, it appears that you should reduce the Tree Support Branch Angle setting. That will make the angle you describe less "extreme".

Enverex, 2019-02-18:
This may have been an odd setting hidden away somewhere that I'd not reset and didn't realise because setting the angle to the default 40 still had the same issue.

I've cleared everything and reset my profiles, then loaded it again and now I'm getting the same as your image. I can't even recreate my original issue with any combination I've tried so far, so god knows which option was actually causing that.

Looks like this was a weird combination of options rather than a bug, sorry about that.

closed 2019-02-18 by Enverex