Ultimaker/Cura #5318: Support haning lose

Perohlsson123, 2019-02-16:
**Application Version**
Beta 4.0.0

**Platform**
Windows 10

**Printer**
Ultimaker S5

**Steps to Reproduce**
Slicing with "enable conical support" enabeld, Conical support settings is 35 degrees and 5mm.

**Actual Results**
See atteched file

**Expected results**
it should not hang lose, becuse its not that complex. 

**Additional Information**
Atteched Picture:
[image]

Ghostkeeper, 2019-02-18:
I'm not able to reproduce this using my test models. Can you provide us with a project file that reproduces the problem?

Perohlsson123, 2019-02-18:
Hi

I cant get it to be unsupported again, and im using the same profile and part???.

But now im having another unsupported issue. See attached Picture.

Attached is:
Profile:     2. EUPart, Normal, V00
Part:         33880-00_R0_3DP

Best regards
Per Ohlsson

Phone: +46 8 620 20 94
Mobile: +46 72 571 35 94
per.ohlsson@georgfischer.com<mailto:per.ohlsson@georgfischer.com>

production Engineer
System 3R International AB
Sorterargatan 1
162 50 Vällingby
Sweden
www.system3r.com<http://www.system3r.com/>

This e-mail (including attachments) is confidential and may be privileged. Please delete if obtained in error and email confirmation to the sender. Thank you for your cooperation.

Ghostkeeper, 2019-02-18:
Attachments don't work via the e-mail reply feature of Github. You have to actually go on the site and upload it there.

Perohlsson123, 2019-02-18:
Ok

Trying with this :)

[Hanging loose.zip](https://github.com/Ultimaker/Cura/files/2875257/Hanging.loose.zip)

closed 2019-02-18 by Perohlsson123

Ghostkeeper, 2019-02-18:
Yeah, I can't reproduce that either with your model. It seems to work fine so far.

[image]

I would really print that upside down, by the way, but maybe you need that top surface to be pretty :P

Ghostkeeper, 2019-02-18:
The conical angles are not visible on the sides because it's already at the minimum thickness for the cones there (the inside is hollow).