Ultimaker/Cura #5319: Support generation under object when not needed

pcollingwood, 2019-02-16:
**Application Version**
CURA 3.6.0

**Platform**
Windows 10

**Printer**
Home made Delta running a Smoothie board

**Steps to Reproduce**
Place model, slice.

**Actual Results**
File is created

**Expected results**
There should be no internal supports, but there are.

**Additional Information**
The internal supports only disappear when support angle is set to 0 degrees.
[coneVEX .curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2872328/coneVEX.curaproject.3mf.zip)
[CURA Delta settings.zip](https://github.com/Ultimaker/Cura/files/2872331/CURA.Delta.settings.zip)

pcollingwood, 2019-02-16:
Correction:
**Additional Information**
The internal supports only disappear when support angle is set to **90** degrees.

Ghostkeeper, 2019-02-18:
This issue is already fixed for 4.0. See https://github.com/Ultimaker/Cura/issues/4583.

closed 2019-02-18 by Ghostkeeper