Ultimaker/Cura #5321: Allow changing retraction speed/length/prime speed in ChangeAtZ

Yannik, 2019-02-17:
This would be very helpful for retraction calibration towers like this:

[image]

Ghostkeeper, 2019-02-18:
We find ChangeAtZ's current implementation to be hard to maintain. So we don't, or hardly, change it. I wouldn't put your hopes up. But if someone does want to tackle this, please make sure that this works together with other features of ChangeAtZ, such as changing the flow (which shouldn't affect the retraction distance).

Venoryk, 2019-07-22:
I second this. I was just attempting to configure a retraction test to get my printer dialed in only to find that I would have to use S3D in order to do it. As i'm somewhat new to 3D Printing Cura has been the only slicer that I've had any experience with. Would love to see this feature added to Cura.

Ghostkeeper, 2020-05-11:
This got implemented. It'll be in the upcoming version!

closed 2020-05-11 by Ghostkeeper

Yannik, 2020-05-11:
@Ghostkeeper Which version will this be in? 4.5?

Ghostkeeper, 2020-05-13:
No, the upcoming version would be 4.7.