Ultimaker/Cura #5322: [3.6.0] Initial layer thickness setting not used

ategen3rt, 2019-02-17:
**Application Version**
3.6.0

**Platform**
Windows

**Printer**
Ender 3

**Steps to Reproduce**
Use file: 
[bad_initial_layer.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2873197/bad_initial_layer.curaproject.3mf.zip)

Start with the draft profile
Change layer height to 0.08
Ensure initial layer height is 0.2
Ensure build plate adhesion is anything other than raft (none or skirt or brim)
Prepare

**Actual Results**
Initial layer is 0.08mm

**Expected results**
Initial Layer is 0.2mm

**Additional Information**
If I use raft, the initial layer height is fine.
Skirt and None have the same problem
Other models work without issue using the same settings

smartavionics, 2019-02-17:
Hello @ategen3rt , I just loaded your project into my Cura which is based on the current master branch and sliced using exactly the same settings. When I look at the gcode, the first layer is at z = 0.2 and the second is at z = 0.28. Perhaps the issue has been fixed in the current Cura?

[image]

Ghostkeeper, 2019-02-18:
I also can't reproduce this in Cura 3.6. I sliced your project file and it produced this g-code: [CE3_bad_initial_layer.zip](https://github.com/Ultimaker/Cura/files/2874576/CE3_bad_initial_layer.zip). I cut off most of the g-code to reduce the upload size for Github (just the first 12 layers are left).

The g-code starts like this:
```
;FLAVOR:Marlin
;TIME:9636
;Filament used: 1.4621m
;Layer height: 0.08
;Generated with Cura_SteamEngine 3.6.0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
G28 ; Home all axes
G29; BLTouch mesh generation
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X2 Y20 Z0.3 F5000.0 ; Move to start position
G1 X2 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X2.3 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X2.3 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G92 E0
G1 F2400 E-5
;LAYER_COUNT:948
;LAYER:0
M107
M204 S5000
M205 X30 Y30
G0 F3600 X110.773 Y128.988 Z0.2
M204 S500
M205 X20 Y20
;TYPE:SKIRT
G1 F2400 E0
G1 F1800 X111.384 Y128.538 E0.0101
```

The `Z0.2` in the `G0` command towards the end sets the initial layer's height. So it appears to work correctly.

ategen3rt, 2019-02-18:
I will take a look when I have access to that computer and upload the first part of the gcode.  I noticed it because the amount of filament extruded for the first layer was so low, I thought the nozzle was clogged.  Other prints ran fine, but this one consistently under extruded.

closed 2020-06-30 by Ghostkeeper