Ultimaker/Cura #5324: Brim only on outside notworking

jeremykoerber, 2019-02-19:
I believe this issue was already filed in 3.x, but closed as "works as intended".  But I'm having problems with brims on the inside in 4.0.0 and thought I'd raise the issue once more. Brim on the outside should mean that the brim is only on the outside of the model just like the tool tip says. I'm getting brims on the inside of my model which are hard to remove because they're being applied to cursive lettering that happens to be under the model, and it's TPU. If that's the "outside", what exactly qualifies as the inside? If this setting doesn't truly cause brims to print on the actual outside of the model, then there should be another setting to really do that. Although that seems silly. Or if I'm just missing something, how do I get rid of the brims shown inside the attached image (the Creative Market text), but keep them on the outside of the phone case? Thanks.
[image]

Ghostkeeper, 2019-02-19:
Duplicate of #4222.

BagelOrb, 2020-07-08:
The biggest issue here is that there is brim on the inside of the outer part, which is technically a different issue from there being brim on the outside of the inner part of the e's and the a.

ToyMaker62, 2020-09-06:
"Brim Only on Outside" selected, but the brim is printing inside of the hinge joint.

[image]

BagelOrb, 2020-09-07:
The way I see it it's on the outside of the inner ring.

Perhaps we should just make it a dropdown: everywhere, surrounded outsides, external outsides only.

rogerlucas-veea, 2021-11-22:
Hi All, this is still an issue with Cura 4.12.1.  If you have a model with nested shapes (in my case, text debossed on the bottom, so a letter "B" has two nested shapes), you get the brim printed around the inner shapes, which is unwanted.

GregValiant, 2026-09-07:
I'm cleaning up these old reports and requests.  I'll go ahead and close this.  If the problem still persists in 5.13.0 then please make out a new report so it goes to the top of the pile again.

closed 2026-09-07 by GregValiant