Ultimaker/Cura #5330: filling error of the second extruder

Goodfeat, 2019-02-19:
Hello, looks like I have found an error that most likely goes from version to version of the program.
If you turn off the first extruder in the slicer and leave only the second, then the filling of the model works fine and changes depending on the percentage. But if the program is closed and opened the next time, the filling section in "custom" stops responding. But at the same time the filling section in "Recomended" still works.
The same happens to the filling pattern.

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

rburema, 2019-02-20:
The settings that can be changed per extruder require that the current extruder is selected. What I think happened is that you only changed the infill setting for the first extruder:

With the first (disabled) extruder selected:
[image]

With the second (enabled) extruder selected:
[image]

This does raise some UX/GUI questions about those options being active, but conversely, what if the user wants to change those options and later enable the extruder?

Can you confirm or deny that this is what happened, or is there something else going on?

cheers,
Remco.

Goodfeat, 2019-02-20:
Filling is changed only in the section "recommended", in "custom" the filling does not respond to the setting.
the second extruder is turned on, the first is turned off and there is no access to it in the settings
[image]
[image]
[image]

Goodfeat, 2019-02-20:
Indeed, there is no such problem in CURA 4.0, but in 3.6 there is

rburema, 2019-02-20:
I see what the problem is now.

I'm sorry, but fixes always go into the latest version. If you want a stable build with this fixed, you'll have to wait until 4.0.0 is out (hopefully somewhere next month, but we'll release 4.0-BETA-2 hopefully next week or thereabouts).

Goodfeat, 2019-02-22:
OK, I will wait for the update
Thanks

closed 2019-02-22 by Ghostkeeper