Ultimaker/Cura #5337: [3.6.0] Layer View slow

upatamby, 2019-02-20:
**Application Version**
3.6.0

**Platform**
Windows 10 Pro,  i5 3,1GHz
Intel HD Graphics 4600

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Load a big model (stl)
Choose Layer View and press Prepare

**Actual Results**
The layer view is created. During creation Cura is not responsive (I understand that). GPU usage is 100%. After it has finished creating the layer view things seem to work again, but Cura than behaves very strange: I can move my mouse cursor over the model without problems, but as soon as I hover over the right panel with the settings, or over the main menu Cura is slow and again uses 100% GPU. Why? There is not much to calculate anymore, because the model is finished. 
If I check Layer View compatibility mode the behaviour when hovering over the menu's is better (not good and still using 60% GPU!!!), but inspecting layers is extremely slow and almost useless.

**Expected results**
Cura should use GPU only as much as necessary (so I don't need to buy a better graphics card, just to be able to navigate through the Cura menu's).

Ghostkeeper, 2019-02-21:
It's rendering the scene again because your mouse cursor might be hovering something that needs to be highlighted. This is both to render the cursor collision areas (invisible areas in the scene that trigger some behaviour when the cursor hits it) and if something needs to change upon hover, like the X/Y/Z coloured tool handles that highlight when hovered.

I agree that we should only need to re-render in the second case, if a collision zone was actually hit by the mouse.

Ghostkeeper, 2021-06-28:
This improvement was implemented in Cura 4.7.

closed 2021-06-28 by Ghostkeeper