Ultimaker/Cura #5342: Ability to set ironing layer height

kyjak, 2019-02-22:
**Application Version**
CURA 3.6

**Platform**
Win 10

**Printer**
MK3

**Steps to Reproduce**
Print test cube with ironing on, 15% iron flow

**Actual Results**
The ironed layer does not have the correct height, it adds more to the model than expected

**Expected results**
If we set a new variable to compensate for ironed layer thickness, the top layer should be printed
a bit less thick so the top layer + ironed layer height = layer height.

Ghostkeeper, 2019-02-22:
The ironed layer cannot have a different thickness than the rest of the layers. There is not one ironed layer. If you have a model with more than one top surface, all of these surfaces get ironed. And all of these layers (except the highest) can be ironed partially.

If the ironed parts of a layer can have a different layer thickness than the rest of the layer, that runs a risk of colliding with the nozzle when it travels.

The Z position of the ironing is the exact same height as the rest of the layer. So 15% flow for ironing results in 15% overextrusion. Rather set it to 5%-10% if you're experiencing blobs from overextrusion. This overextrusion is on purpose so that it fills the gaps in between the previously printed lines. Putting the material higher defeats the purpose then.

kyjak, 2019-02-23:
We misunderstand each other. The model has a designed height of 10mm. When I slice it without ironing it gets printed as 10.00mm. When I turn ironing ON, the print has a height of 10.12mm. (Layer height 0,2mm).
So ironing destroys the height dimension quite a lot. My question was if we can somehow compensate for this with a new setting.s

Ghostkeeper, 2019-02-28:
I'm not able to reproduce that then. I sliced a 10mm cube with layer height 0.1mm, and the highest Z coordinate was 9.97mm. When I turned ironing on, it was still 9.97mm.

I don't see how ironing affects the height of layers in any way right now, in our code.

kyjak, 2019-02-28:
It is not the code, the print result is wrong, cause ironing adds layer height and Cura does not know about it, so Cura cannot compensate for it. The best way would be to add an Ironing parameter "height" and Cura should substract that delta from the top layer so after ironing the printed layer has a good height.

Ghostkeeper, 2019-03-01:
Ah, I see what you mean! Sorry. It's actually inaccuracy in the print process due to ironing overextruding.

So what you'd like to see is that we reduce the flow of the part that will get ironed, so that it doesn't overextrude? If you meant it to also reduce the Z height at which to print the skin that will get ironed, that bumps again with our requirement that the layers must be planar.

kyjak, 2019-03-04:
The problem is you need some amount of extrusion during ironing and that amount needed does increase the ironing layer height. It is not zero. So what would be great, is to substract a chosen constant from the last layer height before ironing, make it a bit less tall, see it like adaptive layer height setting kind of. Otherwise, the only way to fix it is to manually make the STL model less tall, which is a bad option, its hacking.

Ghostkeeper, 2019-03-05:
When ironing the top surface, not the entire layer will be top surface, usually. If you have a stair case and are printing one of the lower steps, then part of your layer will be ironed (the step) and part of your layer will not be (the not-step). So if part of the layer is printed at a lower position than the rest of the layer, then your layer is no longer planar and you run the risk of needing to make travel moves through previously-printed parts that are higher.

kyjak, 2019-03-14:
[image]
here is a graphics for the idea. when you do a top layer, you separate paths for iron and non iron faces.
when you do non iron, you extrude full height of layer.
when you do iron, you extrude the layer height minus the iron height added and then do ironing above the same area.
i think there is no obstruction to do so

Ghostkeeper, 2019-03-14:
Let me illustrate something similar to clarify how this can cause the nozzle to collide with previously printed parts:

[image]

If due to the path optimization the yellow part needs to get printed first, then the reduced top layer on the right, then the reduced top layer on the left, then the nozzle will collide with the normal layer and scrape the material off there. This has a danger of pulling the print off the build plate, or causing a layer shift.

kyjak, 2019-03-14:
I see now the problem. What if CURA would split the yellow (whole layer) also into 2 parts.
one reduced layer and one normal non-ironed layer of iron height?