Ultimaker/Cura #5345: [3.6.0] Infill layer thickness gap bug

Levious, 2019-02-22:
**Application Version**
3.6.0

**Platform**
MacOS Mojave 10.14

**Printer**
Custom ULTI

**Steps to Reproduce**
Layer height - 0.05
Infill layer thickness - 0.2

**Actual Results**
Big gap between infill and wall

**Expected results**
No gap

**Additional Information**
I changed the infill overlap from 10% to 80% and the result did not change. For the experiment, I tried the same model on the same settings in the simpliify - everything was fine.
Is this a bug or am I doing something wrong somewhere?
If the thickness of the layer coincides with the thickness of the infill - everything is ok. If not, a gap appears.

10% overlap
[image]

80% overlap
[image]

smartavionics, 2019-02-24:
Hello @Levious, please zip up and attach a project that exhibits this problem. Thanks.

Levious, 2019-02-26:
> Hello @Levious, please zip up and attach a project that exhibits this problem. Thanks.

You can create any simple object of your own like a cube or cylinder and see this bug in yourself. I attach my project. Thank.
[Just 20 box.3mf.zip](https://github.com/Ultimaker/Cura/files/2905142/Just.20.box.3mf.zip)

smartavionics, 2019-02-26:
Hello @Levious , that's not a project file, it only contains the model. Please do File -> Save to create a project file that also contains your settings. Thanks.

Levious, 2019-02-26:
> Hello @Levious , that's not a project file, it only contains the model. Please do File -> Save to create a project file that also contains your settings. Thanks.

[Just 20 box.3mf.zip](https://github.com/Ultimaker/Cura/files/2907455/Just.20.box.3mf.zip)

smartavionics, 2019-02-27:
Hello @Levious , I have fixed this issue for most of the infill types and will submit a PR so that the fix can be incorporated in Ultimaker's Cura sometime in the future. Thanks for the bug report.

smartavionics, 2019-02-27:
[image]

Ghostkeeper, 2019-02-27:
Good find!

Block137, 2019-07-05:
Any temporary workaround for this?

Appears to affect support as well, I have a problem where PVA interface does not adhere to PLA support infill.

closed 2019-10-25 by rburema

Block137, 2019-10-25:
Can we(users) expect this fix to be in next release?

My current workaround is to half the line width and double the flow rate.

smartavionics, 2019-10-25:
Yes, unless some problem is discovered, it will be in 4.4.

You can already benefit from this fix if you are using one of my builds which you can find at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0.

Ghostkeeper, 2019-10-30:
Yes, it's merged for 4.4.