Ultimaker/Cura #5346: [3.6.0] [4.0.0-beta] For same STL, gcode generated in windows 10 works, gcode generated in ubuntu 16.04 does not on Ender 3

xninop, 2019-02-22:
**Application Version**
3.6.0

**Platform**
Ubuntu 16.04 LTS
Windows 10

**Printer**
Creality Ender 3

**Steps to Reproduce**
\- Load the STL, rotate so model lies flat on bed and slice under windows10. Save gcode to file then print via USB on Ender 3.
\- Load the same STL in Cura 3.6.0, rotate the same way as above and slice under Ubuntu16.04. Save gcode to file then print via USB to same printer.
\- Load the same STL in Cura 4.0.0-beta, rotate the same way as above and slice under Ubuntu16.04. Save gcode to file then print via USB to same printer.
\- Load the windows-generated gcode in ubuntu version and print on same printer.

**Actual Results**
\- The print from windows gcode is fine.
\- The print from ubuntu gcode (both cura3.6.0 and cura 4.0.0-beta) is a mess. Actually causes extruder skipping with nozzle seemingly not moving up in Z correctly and scraping into previous layers. Potentially damaging to printer!
\- The print launched from ubuntu using windows-generated gcode is fine.

**Expected results**
Expected a good print in every case

**Additional Information**
The windows10-generated gcode:
https://drive.google.com/open?id=1Pw0oa5qIL5KSWj-lbBXApIjTZUahjTC1

The ubuntu-generated gcode:
https://drive.google.com/open?id=1uHN9Nm6cZOBp_vwiHUg5W2g5XY1NzMf1

smartavionics, 2019-02-22:
Nothing to do with the fact that the are using different gcode flavours?
````
1,2c1,4
< ;FLAVOR:Marlin(Volumetric)
< ;TIME:1413
---
> ;FLAVOR:Marlin
> ;TIME:1414
> ;Filament used: 0.870558m
> ;Layer height: 0.15
````

xninop, 2019-02-22:
I'm new at this so I can't say. Is there a way to change this setting somewhere?

smartavionics, 2019-02-22:
You need to open the machine settings dialog, the gcode flavour is set in there...

[image]

xninop, 2019-02-22:
Thank you smartavionics for quick response. I changed the setting and printed to make sure. It worked!

closed 2019-02-22 by xninop