Ultimaker/Cura #5347: unable to print Body1.stl

dragold, 2019-02-22:
**Application Version**
3.6.0

**Platform**
OS Name	Microsoft Windows 10 Home
Version	10.0.17134 Build 17134
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Manufacturer	Micro-Star International Co., Ltd.
System Model	GP62 7RD
System Type	x64-based PC
System SKU	16J9.3
Processor	Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz, 2801 Mhz, 4 Core(s), 8 Logical Processor(s)

**Printer**
delta bot

**Steps to Reproduce**
1)run Cura administrator

2) open Body1.stl 
[body1.zip](https://github.com/Ultimaker/Cura/files/2895393/body1.zip)

3) rotate:
[image]

1Ex) print setup :
[image]
[image]
[image]
[image]
[image]

4) select : layer view

**Actual Results**
the inside ribs/frame like feature has missing layers
[image]

**Expected results**
the inside ribs/frame like feature shoul'd have missing layers

**Additional Information**
N.A

smartavionics, 2019-02-22:
It's because those walls that are missing are too thin. Try enabling the Print Thin Walls option (YMMV).

dragold, 2019-02-22:
> It's because those walls that are missing are too thin. Try enabling the Print Thin Walls option (YMMV).

dragold, 2019-02-22:
No, it is no work 
[image]

yes I checked in Cad the dimension of that particular feature is 0.5mm

I tried it on slic3r it workes on that

smartavionics, 2019-02-22:
Cura handles thin walls poorly.

dragold, 2019-02-22:
not exactly I believe, I have done thin wall before even without print thin wall 
i

Ghostkeeper, 2019-02-28:
What if you set the line width to 0.49mm? Rounding coordinates down to micrometers can cause walls to disappear even if they are exactly the line width.

dragold, 2019-02-28:
But I say that is not the problem 
if that was the case then cura will also not be able to possess the model, vertically with near no problem
[image]

still, there is another problem(I do not know ) why this is coming (this was also found on possessing the model horizontaly)
[image]

Moreover, if cura cannot handle 0.4mm where it is the standard nozzle size for almost all 3d printers then it's sad....

Anyway, can u tell me how cura handles mesh ??

Ghostkeeper, 2019-03-01:
> if that was the case then cura will also not be able to possess the model, vertically with near no problem

When you orient your mesh vertically, the curve of your surface will cause each slice to be slightly wider than the pure perpendicular thickness (because the surface is sheared w.r.t. the horizontal plane), allowing Cura to fit a line in there.

> still, there is another problem(I do not know ) why this is coming (this was also found on possessing the model horizontally)

What you see there is small gap filling. This had a few major bugs in 3.5 and 3.6 which is improved in 4.0. It's not perfect yet in 4.0 either though, but that is known. I would not expect that to happen outside the outer walls though, unless you configured the slice to have no walls at all...

> Moreover, if cura cannot handle 0.4mm where it is the standard nozzle size for almost all 3d printers then it's sad...

The problem is not 0.4mm specifically. The problem is your line width being exactly the same as the width of your model, and then rounding errors causing the model to be actually 0.003mm thinner in some places than in others.

> Anyway, can u tell me how cura handles mesh ??

Yes, that's documented here: https://github.com/Ultimaker/CuraEngine/blob/master/docs/pipeline.md

dragold, 2019-03-01:
ok...
I have gone throw CuraEngine/docs/pipeline.md and I may have found a way to kill two bird problems with one short but you will have to add 2 stages Post- Generating Areas and do some modes to Generating Areas It is simple by the way.

care to look at it?

Ghostkeeper, 2019-03-05:
What do you need added, then?

dragold, 2019-03-05:
the prosses work as given below :

### 1)Pre-Slicing(Added)

Note: (x1,x2), (x2,x3), (x3,x1) are the planes and x1, x2, x3 are the axis of a body(more no of the axis can be used for better resolution of the object )

generation of 2D layers(Slicing) in terms of (x1,x2) plane =g1 phase
generation of 2D layers (Slicing) in terms of (x2,x3) plane =g2 phase
generation of 2D layers (Slicing) in terms of (x3,x1) plane =g3 phase

### 2)2D-CROSSPOINT(Added)

Note: (x1,x2), (x2,x3), (x3,x1) are the planes and x1, x2, x3 are the axis of a body
Note: X, Y, Z  are coordinates of the workspace 

In this stage the set of 2D layers produced  in  g1, g2 and g3 phase where put together in the workspace, hear the Dot-points are created where each line of 2D layers of g1, g2 and g3 intersect creates a Dot- point placed in terms of x1=0,x2=0,x3=0 of the body, that the software can recognise it with respect to X, Y, Z axis in the 3D workspace. 

### 3)Point-Injection(Added)

In this stage, the Dot-points generated in the 2D-CROSSPOINT stage is added into the mesh of object or placed along with the mesh with a common origin with symmetrical axes.

### 4) Final-slicing (Added)

Note: X, Y, Z  are coordinates of the workspace 

In this stage, the object is Sliced for the last time in terms of X, Y, Z (As per user required),
However, when creating the 2D layers Dot-point at which it will make a line by itself along with the line created by using the mesh, A combination of above given will give the software a high-resolution Slice hence the Final-slice 

  
### 5)Generating Areas (Common)

### 6)Generating Paths (Common)

### 7) Inserts (Common)

### 8)G-code (Common)

Ghostkeeper, 2019-03-06:
Uh, I'm going to need some clarification. I looked at that together with @rburema we but couldn't really figure out what it is that you're intending to do.

It sounds like you're doing some sort of principal component analysis where you define 3 arbitrary directions and slice the object in each of the 3 planes. But of course the cross sections there are 2D polygons in a 3D plane which will not collide exactly on the plane except in edge cases (however common) so you will not find many intersections of the g1, g2 and g3 polygons.

And then you take these points of intersection and add them to the mesh. But how that is "added" is unclear to us. Are you trying to use them to make a sort of voxelised representation of the mesh? Or maybe you're trying to perform a 3D offset?

dragold, 2019-03-07:
In simple words, we are slicing the object in 3 arbitrary directions then using that 3 sets of slice combining them to create points where the lines of the set of slice intersect(we are not using any principal component analysis )from this we can get an infinite number of points  but we can select a set  of points in the area which is needed to make a voxelized representation of the mesh in the selected area in the mesh but instead of making a new mesh from this, we go to directly do to 2D slicing where software will play connect the dots with the point while also slicing the mesh

``It sounds like you're doing some sort of principal component analysis where you define 3 arbitrary directions and slice the object in each of the 3 planes. But of course the cross sections there are 2D polygons in a 3D plane which will not collide exactly on the plane except in edge cases (however common) so you will not find many intersections of the g1, g2 and g3 polygons''

Note: g1, g2, g3 are not polygons comment a day ago  they are sets of sliced images.... we are not talking polygons but the lines of the image of the cross-section
 
I am pretty sure that if u are given the function of sides of the plane as a line

GregValiant, 2026-09-07:
I'm cleaning up these old reports and requests.  I'll go ahead and close this.  If the problem still persists in 5.13.0 then please make out a new report so it goes to the top of the pile again.

closed 2026-09-07 by GregValiant