Ultimaker/Cura #5348: [4.0.0 ] Freeze/Hangs on splash screen

Deneteus, 2019-02-22:
**Application Version**
4.0.0

**Platform**
Windows 10 x64

[image]

[image]

**Printer**
No printer is connected to the PC but it is set to Prusa Mk3.
I use Octoprint normally but this is a fresh install of 4.0
 
**Steps to Reproduce**
Opening the application causes it to load and freeze on splash screen.

**Expected results**
The application should not freeze on loading.

**Additional Information**
Version 3.6 was working fine. 
Video of issue located here:  https://youtu.be/aA4oXi092uM

Deneteus, 2019-02-23:
After an extremely long time (hour and a half) I came back to the application open on the desktop even though the window had disappeared on the task bar. Clicking on Preferences and Materials and it Freezes up.

Ghostkeeper, 2019-02-28:
Nothing in the logs that is relevant to this :disappointed: 

[image]

We've been getting similar reports via Reddit.

Ghostkeeper, 2019-02-28:
Could you try going to that same folder where you found the log, and editing the `cura.cfg` file with a text editor (while Cura is closed). Remove the preferences there that start with `window_` under the `[general]` category. Then start Cura, to see if the problem is fixed?

I'm wondering if this bug occurs when the preferences say to create a window which is smaller than the minimum size of the new GUI.

Deneteus, 2019-03-02:
@Ghostkeeper 

I used Sublime Text and searched for all the occurrences of 'window_' and found only a couple lines to delete. The only thing left was Window_State = 2. When I opened Cura 4 it still froze at 'Loading Interface...'

Ghostkeeper, 2019-03-04:
We're still unable to reproduce this freeze so far, at HQ :disappointed: Even with that .cfg file.

AlanWhereAreYou, 2019-03-19:
I am getting a similar problem: Typically Cura launches on my laptop in less than 30 seconds.

Recently though, sometimes the launch does not go smoothly:
- Sometimes it hangs in the background, no splash screen.
- Sometimes it gets a splash screen and correctly shows progress. In this case, you can see that it's progressing through the load steps but they are taking extremely long - over 10 minutes to launch.

Other notes of questionable accuracy:

- There appears to be a pattern where on first launch after reboot (even if the reboot was hours ago) Cura will launch properly. Then, at some point, it stops being happy and launches after that are slow.
- Usually killing the app and restarting doesn't help. Last time, though, I killed the app and shutdown my NIC, then relaunched at it seemed to be faster. 

I uninstalled and reinstalled. That didn't fix anything.
I'm Cura 3.6.0 up to date as of today on Win10.
I am going to try to grab log files etc.. next time it happens.

AlanWhereAreYou, 2019-03-31:
FYI, my problem turned out to be the "Frozen Path" issue described here: https://github.com/Ultimaker/Cura/issues/5534#issuecomment-478353058

McNeight, 2019-05-29:
This issue was the first hit in a search for 'cura 4.0 hangs' and for me, it was resolved by installing Cura 4.1 with the fix from #5534 in it.

Ghostkeeper, 2019-05-29:
That makes sense. Thank you for confirming.

closed 2019-05-29 by Ghostkeeper