Ultimaker/Cura #5350: error:slicing time out

JockJo, 2019-02-23:
**Application Version**

Cura: 3.6
UM: 3.6
python: 3.6.5
libarcus: master
curaengine: master
**Platform**

windows 10
**Printer**

Ultimaker S5
**Steps to Reproduce and actual results**

i ran cura thought source and  loaded a stl model file (123kB). After that, i clicked the prepare button. the program started to slice but  it was still in slicing. By the way, i repeated successfully that i can slice successfully in cuar 3.6 that i download from the ultimaker cura websit(https://ultimaker.com/chs).

**Additional Information**

there are the output. 
```
D:\Users\JockJo\AppData\Local\Programs\Python\Python36\python.exe "D:/documents/Cura/cura 3.6/Cura/cura_app.py"
[MainThread] UM.Resources.__initializeStoragePaths [414]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [425]: Config storage path is C:\Users\JockJo\AppData\Roaming\cura\master
[MainThread] UM.Resources.__initializeStoragePaths [433]: Data storage path is C:\Users\JockJo\AppData\Roaming\cura\master
[MainThread] UM.Resources.__initializeStoragePaths [445]: Cache storage path is C:\Users\JockJo\AppData\Local\cura\master\cache
[MainThread] UM.PackageManager.__init__ [49]: Found bundled packages JSON file: D:\Cura\Cura\Uranium\UM\..\resources\bundled_packages\uranium.json
[MainThread] UM.PackageManager.__init__ [49]: Found bundled packages JSON file: D:\documents\Cura\cura 3.6\Cura\cura\..\resources\bundled_packages\cura.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [138]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [144]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [148]: Initializing version upgrade manager ...
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_definition_changes] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_variant] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_material] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_quality] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [empty_quality_changes] added.
[MainThread] UM.PackageManager._loadManagementData [83]: Loaded bundled packages data from D:\Cura\Cura\Uranium\UM\..\resources\bundled_packages\uranium.json
[MainThread] UM.PackageManager._loadManagementData [83]: Loaded bundled packages data from D:\documents\Cura\cura 3.6\Cura\cura\..\resources\bundled_packages\cura.json
[MainThread] UM.PackageManager._loadManagementData [105]: Loaded user packages management file from C:\Users\JockJo\AppData\Roaming\cura\master\packages.json
[MainThread] UM.PackageManager._saveManagementData [117]: Package management file C:\Users\JockJo\AppData\Roaming\cura\master\packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [92]: Loading plugin configuration file 'C:\Users\JockJo\AppData\Roaming\cura\master\plugins.json'
2019-02-23 17:39:19,754 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ConsoleLogger
2019-02-23 17:39:19,793 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [80]: Found CuraEngine at: CuraEngine.exe
2019-02-23 17:39:19,801 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraEngineBackend
2019-02-23 17:39:19,817 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin ConsoleLogger was already loaded
2019-02-23 17:39:19,834 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJReader
2019-02-23 17:39:19,854 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJWriter
2019-02-23 17:39:19,882 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLReader
2019-02-23 17:39:19,902 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLWriter
2019-02-23 17:39:19,919 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FileLogger
2019-02-23 17:39:19,937 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalContainerProvider
2019-02-23 17:39:19,960 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalFileOutputDevice
2019-02-23 17:39:19,980 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CameraTool
2019-02-23 17:39:20,004 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MirrorTool
2019-02-23 17:39:20,029 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RotateTool
2019-02-23 17:39:20,056 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ScaleTool
2019-02-23 17:39:20,077 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SelectionTool
2019-02-23 17:39:20,105 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin TranslateTool
2019-02-23 17:39:20,128 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UpdateChecker
2019-02-23 17:39:20,129 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-02-23 17:39:20,150 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimpleView
2019-02-23 17:39:20,192 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin 3MFReader
2019-02-23 17:39:20,221 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin 3MFWriter
2019-02-23 17:39:20,246 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ChangeLogPlugin
2019-02-23 17:39:20,247 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [378]: Plugin CuraEngineBackend was already loaded
2019-02-23 17:39:20,268 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraProfileReader
2019-02-23 17:39:20,291 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraProfileWriter
2019-02-23 17:39:20,318 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdateChecker
[... 218 lines cut ...]
2019-02-23 17:39:28,156 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\documents\Cura\cura 3.6\Cura\cura\..\resources\shaders\overhang.shader]...
2019-02-23 17:39:28,163 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\documents\Cura\cura 3.6\Cura\cura\..\resources\shaders\striped.shader]...
2019-02-23 17:39:28,174 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\documents\Cura\cura 3.6\Cura\cura\..\resources\shaders\transparent_object.shader]...
2019-02-23 17:39:28,192 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\documents\Cura\cura 3.6\Cura\cura\..\resources\shaders\striped.shader]...
2019-02-23 17:39:28,204 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura\Cura\Uranium\UM\..\resources\shaders\default.shader]...
2019-02-23 17:39:28,208 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\documents\Cura\cura 3.6\Cura\cura\..\resources\shaders\grid.shader]...
2019-02-23 17:39:28,213 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura\Cura\Uranium\UM\..\resources\shaders\platform.shader]...
2019-02-23 17:39:28,348 - DEBUG - [MainThread] cura.Machines.QualityManager.initialize [136]: Lookup tables updated.
2019-02-23 17:39:28,349 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-02-23 17:39:28,350 - INFO - [MainThread] cura.Machines.QualityManager.getQualityChangesGroups [179]: Cannot find node for machine def [ultimaker_s5] in QualityChanges lookup table
2019-02-23 17:39:28,351 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [50]: Updating QualityProfilesDropDownMenuModel.
2019-02-23 17:39:29,935 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 2.2s
2019-02-23 17:39:38,377 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
[... cut ...]
2019-02-23 17:39:56,487 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [227]: Starting to slice...
2019-02-23 17:39:56,489 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [244]: Going to slice build plate [0]!
2019-02-23 17:39:57,741 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [430]: Sending slice message took 1.2522835731506348 seconds
2019-02-23 17:42:44,052 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
```

nallath, 2019-02-25:
But if you're slicing with Cura 3.6, why are you using the master version of the engine? You should probably use the 3.6 version of the engine as well.

Ghostkeeper, 2019-02-25:
I would expect the engine to crash because there is some new setting that the engine uses that your 3.6 version of the front-end doesn't produce.

closed 2019-02-25 by Ghostkeeper

JockJo, 2019-02-25:
thanks, i had solved the problem  when i copied some dll file related with curaEngine.