Ultimaker/Cura #5352: [3.6.0] Confused travel with slotted part

FlailAway, 2019-02-23:
Thank you for WRITING Cura!

**Application Version**
3.6.0 Appimage

**Platform**
Mint 19.1

**Printer**
Anycubic Delta Plus

**Steps to Reproduce**
Slice and print

**Actual Results**
It slices fine. It prints little sections at a time swapping from side to side of a slot. See travel moves in image. It does about 5 lines one side of the slot then scoots to the other side, does five lines there then back to the first side. Almost seems to be mirroring the moves. Same for top, bottom and infill.

**Expected results**
Fill one side of the slot then work around to fill it all.
 
**Additional Information**
[image]

smartavionics, 2019-02-23:
The reason it is doing those travels is because the infill print order logic uses as-the-crow-flies distances by default. If you enable the experimental Infill Travel Optimisation option, it should use the combing distances instead which should reduce the bouncing backwards and forwards you are seeing there.

FlailAway, 2019-02-24:
OK, thanks, that has made quite a difference. Not 100% prefect as it leaves some small areas until last. 

When it does the infill from say, Right to Left, it leaves **an open** triangle at the extreme Right corner and comes back and does that last. 

When it goes from Left to Right for the next layer, it **fills in** that small triangle on the extreme Left corner first, then goes off to do the major infill.

Certainly not enough to complain about, just mentioning it. The "Infill Travel Optimization" made a huge difference; 15 minutes faster in print time.

smartavionics, 2019-02-24:
> When it does the infill from say, Right to Left, it leaves **an open** triangle at the extreme Right corner and comes back and does that last.

> When it goes from Left to Right for the next layer, it first, **fills in** that small triangle on the extreme Left corner, then goes off to do the major infill.

I think I would need to have the project file (or the gcode) to understand what you are saying there.

Ghostkeeper, 2019-02-25:
The way that Cura optimises the order in which to draw lines is as such:

1. Iterate over all endpoints of all not-yet-printed lines to find which is the closest one.
2. Travel to the closest endpoint of that line.
3. Print the line.
4. Repeat from step 1.

Finding the optimal order (shortest travel moves) is an instance of the [Travelling Salesman Problem](https://en.wikipedia.org/wiki/Traveling_Salesman_Problem) for which all known solutions are exponential due to its combinatoric nature. The current algorithm described above is called the [Nearest Neighbour algorithm](https://en.wikipedia.org/wiki/Nearest_neighbour_algorithm) which is a very simpler one. Several years ago we tried implementing the Random Insertion method which is a bit more advanced. It yielded a 0%-1% printing time reduction (depending on the model) so we didn't find it worth the risk to merge it.