Ultimaker/Cura #5353: [3.6.0] Extra infill walls also adds to outer walls.

FlailAway, 2019-02-23:
I like the idea of the Extra infill walls to use around holes that are going to have bolts through to prevent them crushing up. But,I think we may not be quite there yet. :)

**Application Version**
3.6.0 appimage

**Platform**
Mint 19.1

**Printer**
Any

**Steps to Reproduce**
Slice

**Actual Results**
With 3 Outer walls and defining 3 Extra infill wall, there is no difference to setting the Outer wall count to 6 and not using the Extra infill walls.

**Expected results**
I am thinking that Extra infill walls should just add to all the inner walls.

**Additional Information**
None

smartavionics, 2019-02-23:
> With 3 Outer walls and defining 3 Extra infill wall, there is no difference to setting the Outer wall count to 6 and not using the Extra infill walls.

Actually, there is a difference in that skins will go on top of the extra infill walls but not the normal walls. Here's an image that shows how the skin goes right up to the outside wall and over the infill wall.

[image]

Without the infill wall, that skin would have no support at all in some places where it meets the outer wall.

smartavionics, 2019-02-23:
BTW, what I do for reinforcing around holes is to use an infill mesh with the infill density set to something like 90%.

[image]

[image]

FlailAway, 2019-02-24:
Thanks, I had tried using the infill mesh but it is a struggle when you have a lot of holes to do. Plus it seems almost impossible to aling the mesh with holes not on a "standard" (30/60/90 etc) plane.

Plus, if I make just a single change to the .stl in CAD, I have to do it all again. if there was an "Extra infill walls," that worked just on island-type objects, that would be prefect and no extra work for me.

smartavionics, 2019-02-24:
> Thanks, I had tried using the infill mesh but it is a struggle when you have a lot of holes to do. Plus it seems almost impossible to aling the mesh with holes not on a "standard" (30/60/90 etc) plane.

Understood.

> Plus, if I make just a single change to the .stl in CAD, I have to do it all again. if there was an "Extra infill walls," that worked just on island-type objects, that would be prefect and no extra work for me.

Now there's a thought, how about increasing the wall count just for holes? I might just have a go at that.

FlailAway, 2019-02-25:
> Now there's a thought, how about increasing the wall count just for holes? I might just have a go at that.

If you were able to add an "Wall-Count for Holes," In gratitude, I would be sending you my first-born and maybe even the second. Well, after I am married of course. :)

smartavionics, 2019-02-25:
I work on Cura for free, so no fees would be chargeable! However, not sure at the moment how easy it would be to implement. Will continue to think about it.

Ghostkeeper, 2019-02-25:
To steer a little bit: To get an extra perimeter around all holes, you could find all holes, offset them by a certain wall count times the line width, then add that perimeter to the wall centrelines, offset it again by half a line width and subtract that from the remaining area for infill/skin.

smartavionics, 2019-02-25:
Indeed, I was pondering something along those lines. Thanks.

FlailAway, 2019-02-26:
> Indeed, I was pondering something along those lines. Thanks.

Give a holler if you need some testing done. I have Ubuntu and Mint-19.1/Cinnamon PCs for the 3D printing so Appimage would be good.