Ultimaker/Cura #5358: [3.6.0] Inconsistent behavior when opening .stl, sometimes crashes (createConvexHull)

sealj553, 2019-02-25:
**Application Version**
````
~ $ cura --version
qt5ct: using qt5ct plugin
cura version: 3.6.0
````
````
~ $ python --version
Python 3.7.2
````

**Platform**
Arch Linux
Package installed: (cura 3.6.0-2) https://www.archlinux.org/packages/community/any/cura/

**Steps to Reproduce**
- open Cura
- click "Open File" icon or File->Open File
- select model (3DBenchy.stl)

**Actual Results**
1 of 3 things happen:
- model loads successfully
- Cura crashes
- model loads with distorted vertices and/or model size is (reported to be) astronomically large]

The errors seemingly happen at random.

**Expected results**
Model will open

**Additional Information**
It looks like it always has issues when I see this line:
`WARNING - [MainThread] UM.Mesh.MeshData.createConvexHull [410]: Loaded model is low-dimensional, apply QJ to make it full dimensional`

[log1.txt](https://github.com/Ultimaker/Cura/files/2902340/log1.txt): Open file, delete, open file, delete, open file -> weird vertices [image]

[log2.txt](https://github.com/Ultimaker/Cura/files/2902358/log2.txt): Open file -> distorted vertices + huge size [image]

[log3.txt](https://github.com/Ultimaker/Cura/files/2902375/log3.txt): Open file -> crash

[3DBenchy.stl.txt](https://github.com/Ultimaker/Cura/files/2902233/3DBenchy.stl.txt)

rburema, 2019-02-27:
Hi @sealj553,
Does this also happen in the 'official'/Ultimaker distribution? We use AppImage for the Linux distribution, so that _should_ also work on Arch. I ask this because for some of our dependencies we have quite specific version restrictions (it's difficult to switch Qt versions for various reasons for instance) and from what I see this isn't always the same for the community package that you run from.
cheers, Remco.

sealj553, 2019-02-28:
I just tried the AppImage and it's working fine. It definitely sounds like a dependency issue then.
Thanks for the help

closed 2019-02-28 by sealj553