Ultimaker/Cura #5359: Filament change plugin doesn't work with Ender 3

RockyTV, 2019-02-25:
**Application Version**
3.6.0

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Steps to Reproduce**
Add the filament change plugin and slice your model again.

**Actual Results**
There's no filament change instruction in the output gcode

**Expected results**
M600 or similar to ask the printer to change filament

**Additional Information**
[cura.log](https://github.com/Ultimaker/Cura/files/2902929/cura.log)

Seems like it can't find the specified layer even though the model has 50 layers. 
[Jail_Lid.stl (rename to `.stl`)](https://github.com/Ultimaker/Cura/files/2902931/Jail_Lid.stl.zip) taken from https://www.thingiverse.com/thing:2244984

RockyTV, 2019-02-25:
For some reason, in that model, layer 45 doesn't exist for the plugin even though it's in the output gcode. Fixed by increasing the layer number by 1, to 47 (specified layer was 46).

rburema, 2019-02-27:
Hi @RockyTV,
Please report any issues with this plugin to the author of that plugin.
In this case, they seem quite active, the top of [the comments page](https://www.thingiverse.com/thing:2077884/comments) suggest a reply within 1 or 2 days. At least I think that's the one you mean?
cheers, Remco.

GregValiant, 2023-09-14:
Can this be closed?  Pause at Height has been adjusted a couple of times over the years.

Xinayder, 2023-09-16:
I think it can be closed, haven't visited this in a while.

closed 2023-09-16 by Xinayder