Ultimaker/Cura #5361: Window is not maximizable or adjustable.

forevertheuni, 2019-02-26:
The windows is immutable and I can't maximize or adjust its size. Only way to move around is to use host+left click to move the windows around.

Screenshot example of my issue at:
https://imgur.com/a/X6F3BOX

**Application Version**
4.0

**Platform**
Ubuntu 18.10

**Printer**
Not relevant

**Steps to Reproduce**
Just open the app

**Actual Results**
Window is too big and not resizable 

**Additional Information**
My screen is 1366x768 resolution
I didn't have this problem in 3.x versions of cura.

Ghostkeeper, 2019-02-26:
After the redesign, we've had to increase the minimum window size. But we've been fiddling with it for a while. You could try again with the second beta but I think the stable version will have it even more optimised.

As a workaround, reducing the system font size should allow you to make the screen smaller. But of course that's annoying for the rest of your workflow.

closed 2019-02-26 by Ghostkeeper

forevertheuni, 2019-02-27:
That means I should change the settings somewhere? can you point me to a post of a file to change things? Thank you for the fast answer. 

JM

Ghostkeeper, 2019-02-27:
You can try the theme file: https://github.com/Ultimaker/Cura/blob/1c6cce316a0facd83f0beac5b204489f3120346a/resources/themes/cura-light/theme.json#L434

If you're on Windows, this file will be in your Program Files (or wherever you installed it). Otherwise it'll be packed inside the application which makes it a bit harder to modify.

forevertheuni, 2019-03-04:
Thanks for the help. I'm in Linux. I'll manage. :) thank you for developing
cura.