Ultimaker/Cura #5362: Build and First Start with dark Theme

Passador99, 2019-02-26:
I'm working on a customized version of cura but i need to start with dark theme as the default theme on first startup then the user can change it via the general menu, is there any way to do that?

fieldOfView, 2019-02-26:
The default_theme is hardcoded here:

https://github.com/Ultimaker/Cura/blob/efeba48ffb2dc69bc2b5613305b7744ca8a64d4f/cura/CuraApplication.py#L166

It is not configurable in the build-system. You could write a plugin (and include it in your build) that changes the theme preferences (once) before the theme layer is initialised. Blackbelt Cura does that here:

https://github.com/BlackBelt3D/Cura/blob/bb-3.6/plugins/BlackBeltPlugin/BlackBeltPlugin.py#L82

Ghostkeeper, 2019-02-26:
I don't see a good reason to make the default of this preference dynamically set in the build system. It's fine like this.

We'll consider your question answered then.

closed 2019-02-26 by Ghostkeeper