Ultimaker/Cura #5367: Per-material custom gcode fields

comps, 2019-02-27:
**Application Version**
3.6.0 / 4.0 Beta (2018)

**Platform**
Linux (Fedora 29)

Apologies for not strictly following the template, but I feel this feature request is better explained in plain text.

Please implement per-material custom gcode field (similar to extruder start/end gcode) that would be inserted after the printer-specific start gcode and after any material change (ie. dual extrusion). This would allow for per-material setting of

- Linear Advance K-factor (current plugin makes it a print quality setting)
- Absolute `M92 E` value (rather than a flow % to some static reference, in print quality settings)
- Mixing extruder factors, creating "virtual" materials in Cura as a mix of multiple physical ones
- `M92 X Y` compensation for ABS shrinkage (though "Horizontal Expansion" print quality setting may be better)
- etc.

It is fairly unreasonable to expect Cura to support all of these (and others) natively, but a custom per-material gcode field (like ie. Slic3rPE has) is the next best thing.

Alternatively, support for conditional statements in printer or extruder start/end gcode could work too, also allowing additional tuning, ie. `M301`/`M304` PID setting per temperature range.

Ghostkeeper, 2019-02-28:
Essentially you want to store extruder start and end g-codes in the material profile so that they get switched out if you change materials, right?

I don't see how steps per mm in `M92 E` is any better than (or different from) adjusting the flow?

Adjusting these settings is possible when editing the config files manually, but we haven't exposed all settings in the material manager. We may at some point do that, but adjusting that sort of machine setting is an expert-expert level feature so it won't benefit a lot of users.

comps, 2019-02-28:
> Essentially you want to store extruder start and end g-codes in the material profile so that they get switched out if you change materials, right?

My personal use case is just selecting different materials and having their properties set (like a material-specific extension of machine start gcode), not necessarily dual extrusion or extruder switching, but for people with dual extrusion, it would work similarly to extruder start/end, just be material-specific.

> I don't see how steps per mm in `M92 E` is any better than (or different from) adjusting the flow?

`M92 E` and % flow rate are indeed mathematically equivalent, but with `M92 E`, I can tune each material individually to an empirically correct/measured value, whereas with a % flow rate, I have to have some arbitrary constant that the % references. This gets a bit awkward especially if the printer can eat both 2.85mm and 1.75mm filaments (like mine does).
Hence I treat `M92 E` as the empirically correct value and use % flow rate for fine-tuning according to what I _feel_ is the best quality.

> Adjusting these settings per material is possible when editing the config files manually, but we haven't exposed all settings in the material manager. We may at some point do that, but adjusting that sort of machine setting is an expert-expert level feature so it won't benefit a lot of users.

I haven't really found any "custom gcode" feature in fdmmaterial definitions, though it's indeed possible to set regular print settings.

The usefulness is hard to estimate; you already have (awesome!) multi-layer bridging settings that probably only expert-experts will ever touch and some additional scripting support would likely target the same audience.
One could argue it's about the tools, not the use cases - if it wasn't for RepRap Firmware's gcode macro support (that arguably has no good common use case), the E3D toolchanger could have taken much longer to arrive. It would be cool to have powerful scripting in Cura (before layer / after layer / between objects / before z-hop / etc.) with conditional evaluation, like some other slicers have, though it's true it might not be a priority.

Thanks either way!

smartavionics, 2019-02-28:
>  The usefulness is hard to estimate; you already have (awesome!) multi-layer bridging settings that probably only expert-experts will ever touch and some additional scripting support would likely target the same audience.

The reason the bridge settings are so comprehensive is because I felt that there was no one-size-fits-all behaviour that would give good results for different materials/models. I was hoping that Ultimaker and individual users would put the time in to experiment and find what works well. I see no evidence of that happening yet but I live in hope (still).

FWIW, I think your feature request could be useful. It wouldn't require much in the backend to implement. The majority of the effort would be required to add the required text fields into the materials dialog. Probably not much effort for a Python wizard but black magic to me.

fieldOfView, 2019-03-01:
There's a complication with multi-extrusion that will mean this feature is more tricky to implement than you would think. Say you have two materials with different start gcodes specified. Which one should be used? Other settings have the same issue (eg bed temperature), but there it is possible to do a calculation on the different values for different materials in the same print (in the case of bed temperature, the highest of the values is chosen).

Ultimaker will not likely implement features that only work for single extrusion, because a unique selling point of their printers is multi-extrusion.

Ghostkeeper, 2019-03-01:
It is possible for the extruder start and end g-codes though. Not really for the global start and end g-codes, but the extruder codes are saved per-extruder and recovered in the engine per-extruder.

comps, 2019-03-01:
The idea is exactly to have the same thing that extruder start/end gcodes are (run on each tool change), except specified for each material separately. The material gcodes would then presumably be written in a way that allows them to be run repeatedly during the print.
The question would then be whether to run material start/end gcode before or after extruder start/end gcode.

jlsjonas, 2020-03-30:
Would also be nice for per-material PID autotuning values

fieldOfView, 2020-03-30:
This is off-topic, but why would you want different PID values per material?

On-topic: I think changing the entire start script snippet just to change 3 values in a single gcode command is a rather blunt instrument. I would suggest requesting to be able to store PID values per material if that is what you want, instead of start gcode per material.

jlsjonas, 2020-03-30:
@fieldOfView different target temperatures having (slightly) different optimal PID values
So I suppose ideally it would be configurable per temperature (range) instead.

The start gcode just seemed like the easiest strategy since there are several things you might want to customize (in case of the Vertex Delta, tuning the startup sequence for example, certain materials need a bigger _priming dot_)

ernwa, 2020-06-28:
@jlsjonas Ideally one could make the equivalent of linear advance for PID; extruder heat loss is proportional to hot end temperature (minus enclosure temperature), fan speed, and rate of filament extrusion (mm3/sec). All of these could be put into a sort of model-based controller that can automatically compensate for these factors after performing an autotune to estimate the amount each contributes. Such controllers are very common in factories to control temperature of things like chemical reactors, but to my outsider's eyes the rate of firmware development in 3d printers is pretty sluggish so who knows when we'l get there.

Probably the thing that will happen first is getting PID values for a couple temperatures and fan speeds and setting the controller PID by interpolating between them whenever you change temperature or fan speed. 

@smartavionics I appreciate your bridging work; I haven't used the third layer yet and am just starting to tune the second, but my experience so far is that the first layer is like running cables across a valley. It needs to have enough space between stringers to avoid them sticking together and wadding up, and adhesion to the end ledges is critical. I set one or two of my walls to be "infill extra walls" when using sparse infill instead of normal ones to try to encourage the bridge lines to cross over enough plastic to stick for sure. My thinking on the second layer is that it should be aligned with the first but could be much fatter & sloppier to form a raft strong enough to avoid lateral deformation when the third layer floor is printed in with perpendicular zigzag. Dunno if a third layer is needed but we shall see.

GregValiant, 2026-09-07:
I'm cleaning up these old reports.  I'll go ahead and close this.
Cura is now capable of performing logic functions in the StartUp/Ending Gcodes.  There are now settings for XY shrinkage and Z shrinkage.

```
{if 'TPU' in material_type}
M92 E119                   ; Steps/mm for TPU
{elif 'PETG CF' in material_type}
M92 E102                  ; Steps/mm for PETG Carbon Fill
{else}
M92 E97                    ; Steps/mm for PLA
{endif}
```

closed 2026-09-07 by GregValiant