Ultimaker/Cura #5368: Poor file naming protocols lead to bad prints.  Easy to fix

drfredc, 2019-02-27:
**Application Version** All I'm aware of...

Cura's current gcode file naming leads to unreliable printing experiences is not flat out bad prints.  

One can't tell looking at a Cura gcode file what nozzle size and material is to be printed.  A printer can have several different nozzle sizes, many of which are incompatible with the gcode of that file which was created with a particular nozzle size.... 

For example, a CCR10S-planter gcode file might be for using the CR10S with a Noz 4, or Noz 8, or whatever...  Printing with PLA, or PETG...  One can't tell from the Cura created file name...  If one tries to print a gcode file with PETG that was PLA gcoded, bad print...  If one tries to print the same gcode for .8 nozzle with a printer using .4 nozzle with either PETG or PLA -- bad print.  Etc, etc...  Seems Cura would want to minimize bad prints due to the simple problem of inadequate file naming protocols.

The solution is relatively simple.  add a customizable file naming section to click boxes for how Cura constructs a print name.   

1- Nozzle size (for active printer), 
2- Material type (PLA, PETG, etc), 
3- Printer name (set in configuration/printers).  
4- Project (name of print file) would seem the four most helpful.  

For print file reliability I'd use file names of  <Nozzle><Material><Project>.  

Some additional options ought to be considered.  One might also offer Material Name.   One would know looking at the file name the gcode is specific to a particular filament where custom filament settings are important for good printing. For example, I've got antimicrobial PLA that is quite brittle and prints a lot better at 225C...  Some PLAs turn to stringy goo at 225C...   Allowing for material name to be included in the gcode file name would help make for more reliable prints as one would know to use a specific filament for that gcode...    

For Dual extruders with dual nozzles and/or one combined nozzle there ought to be additional options, One might also have Material name 1 and Material name 2 for dual color printers.  Perhaps Nozzle 1 and Nozzle 2 if different sized nozzles are considered in dual nozzle gcode creation..  Not sure of the initial need to get this specific.  However, my next printer will be a dual filament printer and I can see how that's going to be an issue... 

Yes, some folks will probably want to be able to customize the order of names in the naming of files...  These nuances could also wait for future additions...  

For print project reliability of Cura gcoded files, just getting the basic 4 mentioned above would be a good starting point...  <nozzle size><print material><PrinterName><ProjectName> 

Currently, I'd probably click three as I currently only have a CR10s, so I don't need printer name.  However, hoping that Cura would use my different printer names when creating file names, , I set up a different name for each nozzle size used in my CR10s (Noz8CR10s for my CR10s with a .8 nozzle, Noz4CR10S, etc)..    However, so far as I know, all versions of Cura seem to ignore these custom names and use the generic CR10s name.  So the current Cura file names ignore the opportunity to help clarify this file reliability problem.   As an ex-programmer, this sort of code addition mentioned above should be simple to do, and the benefit of more reliable prints just from better file naming could be a likely result...

nallath, 2019-02-28:
We've asked what people wanted regarding prefixes for job names a while back on our forum (https://community.ultimaker.com/topic/23152-ultimaker-cura-prefix/). As you can see, a lot of different variants have been suggested. The issue to actually do something with it is still in the backlog. 

That being said, I still feel that the issue you are trying to solve is essentially a firmware problem. The Ultimaker 3 solves this by putting the nozzle sizes & materials used in a header at the start of the g-code. That way the printer can give warnings whenever the active configuration doesn't match the current one. But I can understand that most cheaper printers don't have this functionality, which means that the "band aid" solution is the only way to fix this for everyone.

fieldOfView, 2019-10-08:
I'm working on a plugin that will let you customize the print job prefix:
https://community.ultimaker.com/topic/29734-custom-printjob-prefix-plugin/

fieldOfView, 2021-12-29:
There are now multiple plugins in the Marketplace that implement this.

closed 2021-12-29 by nallath