Ultimaker/Cura #5369: Can't download Cura 2.7.0 installer

szepeu, 2019-02-28:
**Application Version**
2.7.0

**Platform**
Windows 7, Google Chrome, Internet Explorer

**Additional information**
[image]

Appesteijn, 2019-02-28:
Just out of interest, why would you want to download a old (unsupported) version of Cura?

szepeu, 2019-02-28:
I was with this software since version 12 (or 13, don't remember exactly) and was always pleased of the gcode quality and nice, customizable GUI, compared to other slicers. However, in later versions of Cura (like, 3.x) the interface became slower on my computer (i5-3210M, 8GB DDR3, 840 EVO, GT650M), slicing speed drastically increased and it became horrible to use. 
Despite some nice features introduced in newer versions, I've decided to stay with an older, unsupported version because of better GUI experience and slicing speed on my machine. I chose version 2.7.0, because `initial layer horizontal expansion` was introduced here, which I use often.

rburema, 2019-02-28:
Hi @szepeu,

The interface will be faster again in 4.0 (compared to the 3.x series at least), since we've introduced a few optimizations (as well as an almost complete UI rework), and speed has our attention. As for the slicing, we're currently looking at a few options to speed it up again, but nothing has been planned/decided yet.

(Some of) The older versions will be downloadable again (but perhaps not from the main site) some time in the future, I think 2.7 included, so stay tuned.

cheers,
Remco.

Ghostkeeper, 2019-03-01:
I have a back-up here: http://dulek.net/work/Cura-2.7.0-win64.exe

We're working on putting these releases back. Someone at UM HQ thought it was a good idea to clean up this folder, but we've seen immediately that people don't like that, especially companies who need to sync the entire company to use the same version everywhere.

diegopradogesto, 2019-03-09:
The installers for the deprecated versions were added back in our server. You can download it again from here: https://ultimaker.com/en/products/ultimaker-cura-software/list

However we encourage you to try newer versions that run faster and have more features ;-)

closed 2019-03-10 by diegopradogesto