Ultimaker/Cura #5371: Cura 3.6.0 shows it's connected to MP Select mini V2, but isn't reading or setting temperature when trying to print

mybaum, 2019-02-28:
**Application Version**
3.6.0

**Platform**
Windows 10 Pro

**Printer**
Monoprice Select Mini V2, with the default config Cura provides for that model

**Steps to Reproduce**
Connect Printer to PC, it shows up but Cura can't control it and doesn't read the Temperature

**Actual Results**
Printer shows as connected, but nozzle Temperature shows as 0°C and the Printer doesn't start Printing after selecting Print to USB

**Expected results**
The printer starts heating and printing after heating enough

**Additional Information**
The Printer does show as connected in Cura, but it doesn't seem like it is

Ghostkeeper, 2019-03-01:
Maybe it detects the wrong baud rate or something? It's impossible for us to debug this problem unless we have that printer model.

Does it work in some other front-end like Pronterface? And what baud rate does it use then? Does the printer respond to g-code via serial connection? What does Cura's log say when you make a connection?

mybaum, 2019-03-01:
Hi, im new to 3d printing so im not shure where to find these things. I tried the 4.0 beta and that actually starts to print but stops after the first 2 layers

Ghostkeeper, 2019-03-04:
You can get Pronterface here: http://www.pronterface.com/

I don't have the opportunity to connect Pronterface to a printer but I hope that they'll make it obvious enough to figure out the baud rate of the printer. Try using that to send some g-code such as M106 to turn the fans on/off.

It's weird that Cura does apparently successfully send g-code to the printer but then stops at some point. Maybe the printer is not responding with proper acknowledgement messages?

mybaum, 2019-03-04:
I may have found the problem. Today, my computer wouldn't notice that the printer was connected, and so I think the previous results may have been caused by a loose connection. I'm closing this issue since the problems seem to have nothing to do with Cura

closed 2019-03-04 by mybaum