Ultimaker/Cura #5372: Cura 3.6 Profile at restart

krz4tw, 2019-02-28:
**Application Version**
<3.6>

**Platform**
<Win7 x64>

**Printer**
<I3 Mega>

**Steps to Reproduce**
<When you start Cura it starts with the settings that were set at the time you closed Cura the last time.
Problem is that I have a custom profile that is shown as being active when I start Cura, yet not the settings of this profile are active but the ones from I last closed Cura

I didnt know that and so far my procedure has been:
1. Start Cura and load STL
2. Seeing that my custom profile is already choosen by default at startup
3. Go into the settings and tweak a few parameters for this particular print (activate Supports for example>
4. Slice, Export GCode, Close Cura, Print

5. Start Cura another day to to a completely different print.
6. My profile is shown as being active (in my profile the supports are disabled)
7. Quick slice and print...
8. Come back hours later to see that it has printed supports which I didnt want in this print.

The example with the supports is pretty obvious because you can see it in slicing preview but when you change some obscure settings they will be changed forever without me even noticing.

**Expected results**
<I want Cura not to startup not with the last settings but with the ones of the default Profile. (desired solution)
My Profile is my "baseline" to which I make minor tweaks for every print if they are needed.
If you, the developers, want to start Cura with the last settings, then you just can´t show the profile as being active at startup, let it say "last settings" or somethings. (better but not desired)
Each and every print I made subtle changes to different parameters over the last weeks always thinking that I only changed this one thing compared to my base profile. When I close, this setting is now saved without me knowing and I change another setting for the next print. I was wondering why my prints were becoming worse and worse over time, its because all those single changes were right for this one print but accumulated over weeks and wouldnt work combined>

Thanks
Marc

Ghostkeeper, 2019-03-01:
We do show a fat star on the profile drop-down in custom mode, and a reset arrow in recommended mode, to indicate that you've got current changes.

In our experience, wiping settings when you close Cura is not what most people want. Many people don't use profiles, or just add different printers and switch between them in order to switch between the types of print they like to make. And to many people, closing and re-starting Cura is no semantic reason to reset your current settings, because they like to prepare multiple prints within one session or prepare one print over several sessions.

nallath, 2019-03-01:
This is a pretty fundamental part of how Cura works (and as far as I know, it's the first time this came up). 

I think that most people actually expect this behavior.

closed 2019-03-01 by nallath