Ultimaker/Cura #5373: Draft Wall Speed?

tablatronix, 2019-02-28:
**Application Version**
3.6.0
**Platform**
osx
**Printer**
monoprice smv2

What speed is used for draft wall speed printing ?
Might be useful to clarify in docs.

I thought it might be wall speed, but when I dropped my first layer speed and felt like it was printing slower...

Ghostkeeper, 2019-03-01:
It's using the speed, acceleration, jerk, line width, layer thickness, flow and fan speed of the skirt or brim. So specifically it's going to use the setting "Skirt/Brim Speed", to answer your question.

tablatronix, 2019-03-01:
So i guess my follow up is should it? Since skirt and brim are single layer, If I use a painfully slow first layer, then my entire wall is slow for entire z height. Seems non ideal, but I do see that it might need to be limited somehow since it is a thin wall.

tablatronix, 2019-03-01:
Ok I see what the actual problem is here, `skirt/brim speed` is hidden if `build plate adhesion` is none.

So it cannot be changed manually

Ghostkeeper, 2019-03-05:
Right, good find! Draft shields aren't used very often, so there's probably some bugs in there still.

That said, I don't see the need to configure the draft shield very carefully. If it's printing a little too fast, it's probably going to print fine. If it's printing a little too slow, it still doesn't take a whole lot of time.

tablatronix, 2019-03-05:
True,  of course I had the first layer speed at 10 for some first layer issues, and I had a full wall, so it was certainly noticeable, but normally it would not be an issue.

Ghostkeeper, 2019-03-05:
You can change the first layer speed separately from the skirt/brim speed by the way.

tablatronix, 2019-03-05:
Well yeah but I could not change it before unless I had a wall and a brim enabled, before your fix.
I guess the workaround is to enable a BPA, change brim speed, then un-enable the BPA, I did not think to try that, as the setting would still be set perhaps.

Ghostkeeper, 2019-03-06:
Then I think we'll close this issue with a "good enough" now. We won't implement being able to tune the draft shield speed separately.

closed 2019-03-06 by Ghostkeeper